sebanc/brunch #372: hdcp not working for android videos

tails101, 2020-07-05:
i think hdcp is an issue
google play video and other android video app stay on a black screen but sound is playing
i m using it on a 4770k with rammus. stable v83 and v80.

sebanc, 2020-07-05:
From what I understand it looks like the lack of drm in dev mode. Have you disabled dev mode by removing "cros_debug" from the kernel command line ?

tails101, 2020-07-05:
need to test let me try

sebanc, 2020-07-05:
Forgot to say you need to powerwash after removing "cros_debug" from the kernel command line

tails101, 2020-07-05:
thx for the help
i have good news and bad news
i did remove the debug and did the powerwash
it disabled the shell so i'm not really happy about that.
and the wash took forever.
anyway.
it fixed the drm but in chrome itself only.
so google play videos work on chrome
sadly it is broken in the android app itself
now when scrubbing the timeline the preview apear
but the video itself stay black and the sound still play.
same for my tv app from orange.
so yea something is broken in android drm.

sebanc, 2020-07-06:
What's your laptop model ?

tails101, 2020-07-06:
It is a desktop. Z87e-itx with an Intel 4770k.

sebanc, 2020-07-06:
Ah ok, as it is a desktop your screen is considered as external and hdcp won't work in Android apps. It's the same behaviour on chromeboxes.

tails101, 2020-07-06:
Damn....any workaround for this?

sebanc, 2020-07-06:
Not that I know, I guess they don't include hdcp support in the android container but that's just a guess.

tails101, 2020-07-06:
welp that is an uninstall for me then.

Crescendo-BLYAT, 2020-07-09:
> thx for the help
> i have good news and bad news
> i did remove the debug and did the powerwash
> it disabled the shell so i'm not really happy about that.
> and the wash took forever.
> anyway.
> it fixed the [...]

it rhymes

tails101, 2020-07-09:
lol so a little bit of news
a friend of me came with an old baytrail tablet
i put on it rammus 83
everything worked well mostly out of the box
it had small issue with the power button
touch work ,tablet mode detection doesn't but it is ok with accessibility we have a touch kb.
no sound issue
and even in debug mode it still read every drm content we threw at it.
so yes it is indeed a issue with primary and secondary screen .
the fix would be to add an option for desktop to make chrome os think the hdmi is a primary screen.
the only issue big issue i had was in gl es 3 and vulkan driver are very bad on it
lots of vertex explosions and missing textures.
but gl es 2 works well.

sebanc, 2020-07-11:
I have no idea how to do that. You can still try to use the "video=...." kernel parameter to modify your display settings.

tails101, 2020-07-14:
sebanc
are you sure every new chrome os boxes suffer of drm issues?
i maybe there is recent chrome box images with drm fixes ?
i do not think they will sell an official chromebox with drm issue.

sebanc, 2020-07-15:
You will find below some references, there are many others (search in google "chromebox netflix black screen"):
https://www.reddit.com/r/chromeos/comments/an2uo0/netflix_black_screen_using_external_screen/
https://www.reddit.com/r/chromeos/comments/cnxztu/netflix_app_does_not_play_on_an_external_monitor/

You can also try to verify this by asking in the "purchasing advice" section of the chromeos discord server.

rayqusa, 2020-08-05:
the blackscreen is happen to me in several youtube videos if I opened it on vanced, if I open from chrome or youtube apps it going well.