sebanc/brunch #373: Thank you for mount_internal_drives!

jhedfors, 2020-07-05:
Thank you for adding this option!

Couple of questions:
1. Will renaming affect the original drive, or is that only a 'view'?
2. Any way that we can "Pin" folders or drives, or is that only for Google Drive items?

Crescendo-BLYAT, 2020-07-06:
Renaming is really rename the drive label (like in Windows).

jhedfors, 2020-07-07:
@Crescendo-BLYAT Thanks! I tried to rename it, but it kept giving the error "Use a shorter name" regardless of the length.  No big deal now that I can use Solid Explorer. :)

dubagunta-rohit, 2020-07-07:
hey... Can you tell me how? and what is solid explorer , is it the android app?
@jhedfors

jhedfors, 2020-07-07:
@dubagunta-rohit - Yes, it is an Android app. Now that it can access the areas outside of the Android "sandbox" it can be a Files app replacement. :)

[https://play.google.com/store/apps/details?id=pl.solidexplorer2&hl=en_US](url)

dubagunta-rohit, 2020-07-08:
@jhedfors Can you tell me how to do it? i searched the entire app.. did not find anywhere

jhedfors, 2020-07-08:
@dubagunta-rohit How to do what? Sorry, I am not sure what you are asking...  
After using the mount_internal_drives boot option, the drives show up in Solid Explorer as well as the Files app. Previously, I was editing fstab, which would allow me to access the my drives in the Files app, but not in Solid Explorer With this boot option they show up in both, and I don't have to mess around with fstab every time I update brunch.

jhedfors, 2020-07-11:
@sebanc Could the mountpoint be changed to match the partition name (ex. "SDA1")? 

I noticed that the shortcut I had made in Solid Explorer no longer worked as the path had changed. This may be an issue with Android or Solid Explorer as the names that it assigns are different.

In the Files app I see:
- External Drive
- External Drive 1
- Externalt Drive 2

In Solid Explorer I see:
- USB Drive 1
- USB Drive 2
- USB Drive 3

jhedfors, 2020-07-12:
Opening separate issue

closed 2020-07-12 by jhedfors