sebanc/brunch #375: Lastest Brunch framework problem?

goldwoods2005, 2020-07-06:
My device Lenovo M900 tiny, skylake CPU I5, rammus recovery v83, Brunch v83 testing latest. Dual boot with win 10
-When updated framework successfull in crosh, it hung up in the boot. So I decided to install right from the beginning and it happened again
I forgot to tell you that the stable brunch version working great.
-Another thing is the mount internal drives option not working.
Thanks for brilliance job.
[image]

sebanc, 2020-07-06:
Hi,

It seems the kernel loaded is still the r81 one, even though the brunch framework installed is r83 which is quite weird.

Could you confirm if r83 brunch boots from a usb flash drive install ?

Could you also post your dual boot grub config ?

goldwoods2005, 2020-07-06:
When booting in the USB, everything seems normal; I can boot direct to the Chrome OS
I did try 2 methods to create the img to dual boot from M900
-Using linux mint; boot loop like picture
-Log in to Chrome Os live usb, and create the img to dst; same problem

The weird thing is the r81 brunch seems working without problem, maybe I will try in another PC if I have time and tell you the result
[image]

sebanc, 2020-07-06:
I really think it is related to your dual boot config, could you post it ?

goldwoods2005, 2020-07-06:
Yes, Sorry for not posting it because of I go out for dinner
-I use grub2win to create the dual boot
-I added the custom code:
<img_part=/dev/sdb5
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img>
And leave the windows boot code default

Thank you.

sebanc, 2020-07-06:
Ok the only thing I can think of that could explain this is you have another image file called chromeos.img at the root of another partition; grub only looks for the first partition on which it finds chromeos.img and loads for the kernel from there but it is not necessarily the right partition.

Is it your case ?

goldwoods2005, 2020-07-07:
Yes, I do have.
I will test it back when I arrive at home. 

Thank you.

closed 2020-07-07 by goldwoods2005