sebanc/brunch #376: Compatibility: Razer Blade 2016 & Lenovo L440. Dual booting questions.

tornadox, 2020-07-06:
Hi @sebanc here's some extra info about compatibility.
Installed "Brunch r83 k4.19 testing 20200704" and latest rammus recovery "13020.87.0
83.0.4103.119" on two more laptops. 
The Lenovo Thinkpad L440 (i5-4200m) everything seems to work fine, android apps & linux, hardware also worked w/o issues. 
On Razer Blade 2016 (i7-6700hq + gtx1060) everything also works fine except the webcam. 
It is a 0bda:579f Realtek camera which seems to be buggy in linux https://www.spinics.net/lists/linux-usb/msg193218.html. I tried rammus and reef without any difference. I get an image, but it's teared and lags.

I also tried the UEFI dualboot option as I like to keep Windows as main OS and I don't like playing with extra partitions. I followed the guide, it works remarkably from WSL2 with the only one issue that it creates the grub entry which doesn't work because it shows mount point as it appears in WSL. This is expected. So just changing the entries from img_part=C: to /dev/nvm0n1p3 worked fine. So those who are doing this from Windows still need to boot any linux distro live cd and write down their partition location.

 There are two ways how to have side by side uefi dualbooting using the windows default esp/efi partition and without installing grub2win or grub from linux distro live cd.
How to perform these actions depends on the users knowledge and preference of tools, it can be done fully manually from Windows using refind guide as base https://www.rodsbooks.com/refind/installing.html or it can be done from linux distro live cd or from WinPE live cd. I prefer a mix of those.

1) take the contents of the brunch chrome os efi partition and place it in the esp partition under a new folder called for example brunch, it should contain these files "bootia32.efi bootx64.efi grub.cfg grubx64.efi MokManager.efi"
The content of grub.cfg should be edited and should contain boot entry as provided by brunch img installer process. I'm not yet sure if we need all files. Afterwards the new boot entry should be created, I used easyuefi tool to do this, but can be done via cli with efibootmgr or in uefi shell. Basically this would allow to choose the bootloader by selecting it when powering on the system and using the functional keys.

2)For those who want a better look and feel might install refind as their main bootloader selector. Installation process is quite similar and is described in the link above. The only difference is that the user would not be required to press a functional key on boot and each time can select the OS to boot in nice UI with a custom theme and icons. The only drawback here is that secure boot works nice with brunch, and I'm still reading how to do it for refind. However if your are not interested in better UX on boot, this step can be skipped.

I have two questions regarding dualboot from img file. 
1) how to use the underlying ntfs partitions besides crosh? They can be mounted in crosh and it allows to edit the grub.cfg files w/o issues. But accessing files would more look like copy pasting them to Downloads folder etc, e.g. no direct access. 
2) Is there a huge performance overhead using such img files? Does it have any impact on chromes os performance which usually runs from slower EMMC storage? I read a few articles that the performance drops get to 50% on kernels older than 4.4.

Lastly, In some previous ticket regarding HP Spectre compatibility I noticed that in some Android games there are graphic glitches, now I'm sure they are related to haswel Intel HD Graphics 4XXX graphics core driver as the issue doesn't appear on i7-6700hq skylake which has a HD 530 graphics core. Wondering if those are present on chromebooks that utilize similar graphics cores.

sebanc, 2020-07-11:
Thanks for the report and tips :)

Regarding Razer Blade 2016, if I am not mistaken the needed patch to support the camera has not been included in the mainline kernel yet. If at some point, it is included in the mainline kernel, I can try to backport it.

Regarding 1), I just tried it on my Surface Go and I can access all partitions (including the one where the image is) but my chromeos image is on an ext4 partition. If you  have the image on a ntfs partition, it is probably an ntfs filesystem limitation.

Regarding 2) I did not personally measure this but several users performed benchmarks between full install and dual boot images. All the results reported so far are that the i/o speed is the exact same.

tornadox, 2020-07-17:
Thanks! 
RE 1) if I mount the partition directly via crosh I can access it there and copy-paste files to downloads folder for example. I think with that option also it's possible to explore content of the ntfs root partition in chrosh but not in Chrome OS and Android file managers. It also mount all recovery partitions. The other bad thing was that I saw a couple of time a "Recovering your devices message" - perhaps related to unclean ntfs. Agree that this has something to do with ntfs.
RE 2) - very nice!

RE 3) - will try and report. Thanks for the advice!

Btw, I got refind to work with secure boot, the instruction are not so straightforward as with brunch, but they worked. For some reason I need to go to uefi and add manually the keys, the keys status changes to custom or user afterwards, with brunch it remains in factory state.

sebanc, 2020-07-18:
Usually "Recovering your devices message" happens because fast startup is enabled in Windows, try disabling it (https://www.asus.com/us/support/FAQ/1031533/)

This could actually also be the cause for your internal drive not to be properly mounted in chromeos.

Unrelated question: I see in the first post that you have WSL2 installed, could you please provide me with the output of "cat /proc/version" in WSL2 ?

tornadox, 2020-07-18:
> Usually "Recovering your devices message" happens because fast startup is enabled in Windows, try disabling it (https://www.asus.com/us/support/FAQ/1031533/)

> This could actually also be the cause for your internal drive not to be properly mounted in chromeos.

I had the same suspicion, thx! I will recheck.

> Unrelated question: I see in the first post that you have WSL2 installed, could you please provide me with the output of "cat /proc/version" in WSL2 ?

tornadox@tornadox-ryzen:~$ cat /proc/version
Linux version 4.19.104-microsoft-standard (oe-user@oe-host) (gcc version 8.2.0 (GCC)) #1 SMP Wed Feb 19 06:37:35 UTC 2020

sebanc, 2020-07-19:
Thanks for the WSL2 kernel version :)

tornadox, 2020-07-19:
No luck with graphics using cave recovery. I tried last available haswell recovery (monroe) but it doesn't have android support.

rayqusa, 2020-08-05:
I am curious how to move .img to Micro SD. where to place /dev/nvm0n1p3?
would you like copy full grub command here?

I am using lenovo thinkpad yoga 260 and with grub2win, the bios forbid me to boot from Micro SD card. if I Install Chrome OS on my SD card, it doesn't recognize my SD card.
it may work with your command, please share it here.
thank you

tornadox, 2020-08-07:
Hi @rayqusa 
In your case you need to install grub on the main storage device ESP partition if that is using UEFI in the same logic as windows have it. (I don't have experience with grub2win or MBR-brunch installation). If needed I can make a screenshot and add a tree later.
The simplest way for me was to mount the ESP partition from windows and place there a new folder with brunch bootloader files (grub and bootx64 etc) in grub.cfg you need to provide the path to you microsd, you can only get that by booting into any linux distro and check with command like fdisk -l, you will get a similar output
[image]
in most case that mmcblk is the sdcard that you need to add.

rayqusa, 2020-08-08:
I surfing in internet and I saw that some bios allow booting from Micro SD, and some have not allow.

asus mostly allow booting from micro SD slot, but lenovo not.

I think this is the problem.
in my case grub2win said /dev/nvm0n1p3 not found