sebanc/brunch #377: Macbook Air 2017 wifi only working with insmod command

gembot2k, 2020-07-06:
Hi 
I try to install ChromeOS in my macbook 2017
everything is ok but wifi cant get to work, I use rammus 83 recovery image and latest brunch framework
I already use options=broadcom_wl in my grub config but still wifi device not found, until I try to run "insmod /lib/modules/broadcom_wl.ko" in the shell 
wifi works fine, battery charge indicator now displayed but I need to run insmod every time I boot the ChromeOS
 
I attach my dmesg file
Thanks
[dmesg.txt](https://github.com/sebanc/brunch/files/4880017/dmesg.txt)

sebanc, 2020-07-06:
Hi,

Thanks for the detailed report, indeed this issue has been introduced in the new release, could you try replacing the below text in /etc/init/udev-trigger.conf ?
`start on stopped udev-trigger-early`
by
`start on stopped udev-trigger-early and started network-services`

b1t0y, 2020-07-08:
> Hi,
> 
> Thanks for the detailed report, indeed this issue has been introduced in the new release, could you try replacing the below text in /etc/init/udev-trigger.conf ?
> `start on stopped udev-trigger-early`
> by
> `start on stopped udev-trigger-early and started network-services`

This happened to one of my laptop with broadcom wifi. I did edited the udev-trigger.conf as above and it worked.  My question do we need to do this now on every update of brunch? Or is there going a fix for it? Thanks

sebanc, 2020-07-08:
Thanks for testing it. This fix you tried is going to be included in the next brunch release.

b1t0y, 2020-07-08:
> Thanks for testing it. This fix you tried is going to be included in the next brunch release.

Thank you for your non stop support to make things work for us.  Just fyi,  I have other computer with the same wifi card and not all of them needed the additional started network-services is that behavior normal?  Thanks again.

sebanc, 2020-07-09:
This issue should be fixed in the r83 stable.

Joshagos, 2021-09-17:
WiFi still isn't working on my 2017 MacBook Air.

closed 2026-04-21 by github-actions[bot]