sebanc/brunch #379: Manually adding IPP printer results in reboot of desktop / interface

ericksontx, 2020-07-07:
Using Stumpy with latest Brunch v83 testing (july 4) and v83 recovery. Issue comes up on both Samsung Stumpy Chromeboxes I've tried it with.

In printers when you attempt to manually add an unlisted printer, and do a standard add via putting in the IP address of your printer, all else default (IPP), as soon as you hit OK the screen goes black and the entire Chrome desktop / UI reboots to a fresh start. 

Just a reboot of the interface, not a reboot of the machine or a complete reboot of Chrome.

Crescendo-BLYAT, 2020-07-09:
This is not happens only on specific images.
Mine's Coral with Brunch r83 k4.19 testing 20200704 also experienced the same soft-reboot which after those soft-reboot by just pinning some apps to shelf, other time just by alt+tab (only Spotify android app + Discord web client).

ericksontx, 2020-07-09:
> This is not happens only on specific images.
> Mine's Coral with Brunch r83 k4.19 testing 20200704 sometimes also experienced the same soft-reboot by just pinning some apps to shelf, other time just by alt+tab (only Spotify android app + Discord web client).

I found that this is apparently affecting regular official chrome 83 as well. Tried manually adding a printer via IP address to a fully stock Samsung Chromebook plus running official chrome 83. Same result - instant black screen and reboot of the interface

Crescendo-BLYAT, 2020-07-09:
hmmm interesting, I'll report to the chromium bug tracker....