sebanc/brunch #380: Free HDD Space Not being Recognized

ericksontx, 2020-07-08:
Not sure if this is a simple error on my part. This happens on any version of brunch i've tried. 

When I install to HDD, whether direct to HDD, or as part of a dual boot install (chromeos.img into a partition), Storage Manager only shows the typical 2+ GB free space.

This happens even if its a huge drive being installed to, or  if the partition its being installed to is 100+GB.

Whether I specify a size with the -s variable in chromeos-install, or leave it off, I see no changes in the available space. Same for a full HDD install as for a install into a partition (dual-boot scenario).

In all cases there is no existing content - HDD is wiped clean of partitions, or partition is newly-created and formatted.

Am I missing something? Seems like something simple I'm missing since havent seen other tickets on this and it's been happening to me ever since I began using Brunch prior to 80. Thanks

sebanc, 2020-07-08:
In both cases (dual boot or full HDD install)l it is very weird... Could you provide more details on how you install chromeos ? From which OS and with which exact command ?

jwalcek, 2020-07-10:
Same for me. My partition is 50Gb, yet the actual space I have available is only 8Gb and I'm running low.

I installed it to a secondary 1.5Tb hard drive that already has MX Linux. Using that MX Linux, I followed the steps in your instructions: I download and extracted ChromeOS image and Brunch. I mounted a NTFS formatted 50Gb partition and executed the chrome-install script. I couldn't get the -s option to work no matter what value, so I didn't include anything when running the script. I unmounted the partition and edited grub.

Everything else is working well. It's running on an Ivy Bridge HP laptop.

sebanc, 2020-07-10:
@jwalcek without the "-s" option, the default size of 14 gb is applied so that's normal. Could you provide the command you use to create the image so that I understand why -s option is not working ? also what is the error message displayed ?

CornishRattler, 2020-07-10:
I have exactly the opposite, my storage is showing as 133GB used but the space left is correct. Clean install, no linux enabled

jwalcek, 2020-07-10:
@sebanc the command was "sudo bash chromeos-install.sh -src recoveryimage.bin -dst ~/tmpmount/chromeos.img -s 48"
The error was either "Not enough space to create image file" or another error involving "dd" once I started putting in various numbers to get it to work.

Shamw0w, 2020-07-11:
So I'm not sure if I'm having the same (or a different) problem, so sorry if I'm hijacking the thread, but the amount of storage showing for me doesn't appear correct so I thought it would be best to post here instead of opening a new thread in case it's the same. I installed the OS onto an SD card (60GB), but storage management (in settings) only shows me as having 8GB in total with 6.5GB currently in use (and 1.5GB available). When installing to the SD, I couldn't seem to get it to install to the SD card from my Linux machine (Lubuntu) so I compiled the ISO on my Linux machine, then copied the ISO file over to my Windows computer (Windows 10) using a USB stick, and then installed it from Windows onto the SD card using Rufus.

sebanc, 2020-07-11:
@jwalcek There is a know issue with ext4 partitions where partitioning tools reserve space on partitions which appears free but is not usable. Fix here: https://odzangba.wordpress.com/2010/02/20/how-to-free-reserved-space-on-ext4-partitions/

@Shamw0w What you did is the equivalent of "installing from windows" method, you have to run "sudo resize-data" to get all your storage (backup your important data first just in case)

jwalcek, 2020-07-11:
@sebanc I redid the whole process twice more, and I didn't get the error like I did the first time. So user error on my part I guess.

The second time I chose etx4, however I could not boot into it. The grub error was that it couldn't find the kernel. The third time I did NTFS again, and everything worked. And the storage shows correctly.

So the moral of the story is to use NTFS and follow instructions completely.

Shamw0w, 2020-07-12:
@sebanc , thanks worked a charm and have full capacity now (facepalm)

randomodbuild, 2020-12-08:
I have the same issue, but whenever I try to use "sudo resize-data" it doesn't change anything, and I still only have 8gb of free space despite being on a 500gb hard drive. I used etcher to flash my 15.03 GB chromeos.img image to the hard drive, but the resize-data command still doesn't work. Am I doing something wrong here? One other thing, whenever I do "sudo resize-data", at the very end it does say "The filesystem is already 1263097 (4k) blocks long. Nothing to do!", however it doesn't seem to give me the 500gb of space. I have also tried booting into a live usb and manually resizing the data partition, but then it doesn't boot.

spku-tw, 2021-01-21:
Same issue while install for dual boot in windows via the Brunch r87 stable 20201227.
Two different NB in more then 32 GB partition setup for NTFS, and only got 8GB after installed to LOCAL (step 1 to usb, step 2 to local).

I got the same message "The filesystem is already xxxxxx (4k) blocks long. Nothing to do!" as @randomodbuild 
Still looking for any way to release more space.

obzbdc, 2021-02-15:
> I have the same issue, but whenever I try to use "sudo resize-data" it doesn't change anything, and I still only have 8gb of free space despite being on a 500gb hard drive. I used etcher to flash my [...]

Hi Randomodbuild,

I was having this exact issue having also done an etcher to the hdd install. I fixed it by booting into a Linux Mint USB stick and running Gparted which offered to fix the GPT table. Once that had been done I booted back into ChromeOS and the `sudo resize-data` after that it gave me the full capacity back.

Good luck!

suoko, 2024-10-11:
I installed in an image file, it's rammus 128, and it shows no free nor used space.
Play store does not show apps and Linux install fails.
What's up? Older rammus versions had no problems

[image]
[image]