sebanc/brunch #382: Password-PIN

CornishRattler, 2020-07-09:
Did a search of issues and couldn't find anything related, I've set a PIN to log on instead of my password but it doesn't then give me the option at the lock screen, only password

sebanc, 2020-07-09:
Hi, 

On most chromebooks, PIN only works when resuming from sleep. Only a few chromebooks have PIN sign-in, you can refer to #126

CornishRattler, 2020-07-09:
OK no worries, thanks for the reply

Skawtnyc, 2020-07-09:
This was a minor annoyance for me on the Surface Go. I switched to the Nocturne (Pixel Slate) recovery, but it didn't fix it. However, since I will likely only be using Brunch on my desktop at home, I don't even set the desktop to lock so it's a non-issue for me.

Eventually ChromeOS is going to have face login, rendering this a moot issue.

closed 2026-04-21 by github-actions[bot]