sebanc/brunch #383: Some issues the

divya789vidya, 2020-07-09:
> **Audio Switch not working **

when i connect headphone and  want to use only the microphone of headphone and internal speaker 
the sound doesn't come from the speaker 
that means the audio switch options are not working 
[image]

> **Internal drive mount  option**
 not working with the non dev chromeos i.e. when os start the after mounting the screen gets blanked
and when the  cros_debug is enabled in and mount internal drive is enabled  the Android apps not start it get loading  continuously

> **Bluetooth not working **
the  bluetooth not working in the rtl8723de wireless devices 
[image]

> **acpi_power_button problem **
 i think some the option doesn't work with the  non dev chrome os 
 just as the acpi_power_button  not working with the non dev chrome apps(cros_debug not enabled)
> **Not a big issue **
the android apps takes 3 to 4 min on startup to load  the apps

sebanc, 2020-07-09:
Most of those bugs have been subject to very recent changes. Could you confirm you are using the latest brunch version ?

divya789vidya, 2020-07-10:
yes

divya789vidya, 2020-07-10:
sorry i diint sethe stable release

divya789vidya, 2020-07-10:
going to update

divya789vidya, 2020-07-10:
[dmesg.txt](https://github.com/sebanc/brunch/files/4904620/dmesg.txt)
the Dmesg text info Bluetooth problem

divya789vidya, 2020-07-11:
@sebanc 
today when i enabled the options=mount internal drive option and lot of problem get detected

divya789vidya, 2020-07-11:
> **None of the internal drive is mounted**

[image]

> **Andriod app loading failed**

[image]

after sometime
[image]
[image]

> **Bluetooth Problem**

[image]

DMESG TXT FILE
[dmesg2.txt](https://github.com/sebanc/brunch/files/4906115/dmesg2.txt)

closed 2020-07-11 by sebanc