sebanc/brunch #385: Pen eraser broken on ChromeOS 83 - Surface Go

shinyquagsire23, 2020-07-10:
Updated from rammus+brunch 80 to rammus+brunch 83 using chromeos-update in the cros shell, and now the stylus eraser is failing. Not totally sure what the cause is, however it seems like the eraser events are still occurring in evtest, but Chrome itself and ARC receive all eraser events as if they were just normal pen events.

Unrelated (and possibly deserves its own issue?) but I've also noticed some issues with the keyboard sometimes drops keypresses if it is idled for a moment. There was a similar issue with the touchscreen on the Go which had to be [worked around](https://gist.github.com/shinyquagsire23/291a16f1b5fa3b555f8a9da7692b72ce#file-gistfile1-txt-L49) so I'm guessing it might be similarly fixable (and the eraser issue ~could be related?). I'm trying my old touchscreen fix script with `echo "on" > /sys/devices/pci0000:00/0000:00:14.0/usb1/power/control` added to see if that helps the keyboard drops.

Single eraser tap in evtest:
```
$ evtest /dev/input/event9
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x4f3 product 0x261a version 0x100
Input device name: "ELAN9038:00 04F3:261A Pen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 320 (BTN_TOOL_PEN)
    Event code 321 (BTN_TOOL_RUBBER)
    Event code 330 (BTN_TOUCH)
    Event code 331 (BTN_STYLUS)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   7876
      Min        0
      Max    13312
      Resolution      63
    Event code 1 (ABS_Y)
      Value     40
      Min        0
      Max     8960
      Resolution      64
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max     4096
    Event code 26 (ABS_TILT_X)
      Value      1
      Min    -9000
      Max     9000
      Resolution    5730
    Event code 27 (ABS_TILT_Y)
      Value      1
      Min    -9000
      Max     9000
      Resolution    5730
    Event code 40 (ABS_MISC)
      Value      0
      Min        0
      Max    36000
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1594421910.642928, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1594421910.642928, type 3 (EV_ABS), code 0 (ABS_X), value 10516
Event: time 1594421910.642928, type 3 (EV_ABS), code 1 (ABS_Y), value 5298
Event: time 1594421910.642928, -------------- SYN_REPORT ------------
Event: time 1594421910.657768, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1594421910.657768, -------------- SYN_REPORT ------------
Event: time 1594421910.663360, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1594421910.663360, -------------- SYN_REPORT ------------
Event: time 1594421910.672797, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 1
Event: time 1594421910.672797, -------------- SYN_REPORT ------------
Event: time 1594421910.678524, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1594421910.678524, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1594421910.678524, type 3 (EV_ABS), code 0 (ABS_X), value 10616
Event: time 1594421910.678524, type 3 (EV_ABS), code 1 (ABS_Y), value 5230
Event: time 1594421910.678524, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 1897
Event: time 1594421910.678524, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 5076
Event: time 1594421910.678524, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value 3553
Event: time 1594421910.678524, type 3 (EV_ABS), code 40 (ABS_MISC), value 2900
Event: time 1594421910.678524, -------------- SYN_REPORT ------------
Event: time 1594421910.693362, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2074
Event: time 1594421910.693362, -------------- SYN_REPORT ------------
Event: time 1594421910.702856, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2253
Event: time 1594421910.702856, -------------- SYN_REPORT ------------
Event: time 1594421910.708377, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2376
Event: time 1594421910.708377, -------------- SYN_REPORT ------------
Event: time 1594421910.717859, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2501
Event: time 1594421910.717859, -------------- SYN_REPORT ------------
Event: time 1594421910.723361, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2551
Event: time 1594421910.723361, -------------- SYN_REPORT ------------
Event: time 1594421910.732850, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2602
Event: time 1594421910.732850, -------------- SYN_REPORT ------------
Event: time 1594421910.738334, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 649
Event: time 1594421910.738334, -------------- SYN_REPORT ------------
Event: time 1594421910.753337, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1594421910.753337, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1594421910.753337, type 3 (EV_ABS), code 0 (ABS_X), value 10668
Event: time 1594421910.753337, type 3 (EV_ABS), code 1 (ABS_Y), value 5146
Event: time 1594421910.753337, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1594421910.753337, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 1
Event: time 1594421910.753337, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value 1
Event: time 1594421910.753337, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1594421910.753337, -------------- SYN_REPORT ------------
Event: time 1594421910.762775, type 3 (EV_ABS), code 0 (ABS_X), value 10672
Event: time 1594421910.762775, type 3 (EV_ABS), code 1 (ABS_Y), value 5133
Event: time 1594421910.762775, -------------- SYN_REPORT ------------
Event: time 1594421910.768399, type 3 (EV_ABS), code 0 (ABS_X), value 10676
Event: time 1594421910.768399, type 3 (EV_ABS), code 1 (ABS_Y), value 5117
Event: time 1594421910.768399, -------------- SYN_REPORT ------------
Event: time 1594421910.837444, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1594421910.837444, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 0
Event: time 1594421910.837444, -------------- SYN_REPORT ------------
```

sebanc, 2020-07-13:
Unfortunately, I changed nothing that could impact the stylus eraser function, however the eraser was apparently broken in some libinput versions. I think the chromeos "cmt" driver is based on libinput so it is probably currently based on a version where this function is broken.

I don't think there is something I can do about this for the moment.

shinyquagsire23, 2020-07-17:
Hm, tried updating to beta r84 just to see if it would improve any, it didn't. Looking at it though, it might still be possible to remedy things on the kernel-side? libinput has apparently run into this with other pens, https://gitlab.freedesktop.org/libinput/libinput/-/issues/393 -> https://gitlab.freedesktop.org/libinput/libinput/-/issues/259 

I'm thinking as long as tool mode switches can be made mutually exclusive from the driver end (ie actually compliant with what libinput and everything else seems to expect) then it would probably be fine.

I tried compiling the r83 branch kernel + modules, but for some reason when booting it from GRUB it just resets? no initial logging at all. But I noticed that the latest r83 kernel tree doesn't seem to match up with the latest release. I tried the following patch:
```
```
to alter the behavior when the pen is brought into proximity to (hopefully) unset the pen tool if the eraser ever gets set, and to reset all tools if proximity is left.

shinyquagsire23, 2020-07-31:
Found a kernel which I could compile + boot (https://github.com/kitakar5525/chromeos-kernel-linux-surface) so I could test my theory, seems that it works. New behavior is
```
Event: time 1596217966.758257, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1596217966.758257, type 3 (EV_ABS), code 0 (ABS_X), value 7973
Event: time 1596217966.758257, type 3 (EV_ABS), code 1 (ABS_Y), value 5886
Event: time 1596217966.758257, -------------- SYN_REPORT ------------
Event: time 1596217966.763287, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1596217966.763287, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 1
Event: time 1596217966.763287, -------------- SYN_REPORT ------------
Event: time 1596217966.793404, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1596217966.793404, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1596217966.793404, type 3 (EV_ABS), code 0 (ABS_X), value 8007
Event: time 1596217966.793404, type 3 (EV_ABS), code 1 (ABS_Y), value 5953
Event: time 1596217966.793404, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2688
Event: time 1596217966.793404, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 3858
Event: time 1596217966.793404, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value 3350
Event: time 1596217966.793404, type 3 (EV_ABS), code 40 (ABS_MISC), value 3900
Event: time 1596217966.793404, -------------- SYN_REPORT ------------
Event: time 1596217966.802713, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 1
Event: time 1596217966.802713, -------------- SYN_REPORT ------------
Event: time 1596217966.808256, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 2989
Event: time 1596217966.808256, -------------- SYN_REPORT ------------
Event: time 1596217966.817645, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3290
Event: time 1596217966.817645, -------------- SYN_REPORT ------------
Event: time 1596217966.823278, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3466
Event: time 1596217966.823278, -------------- SYN_REPORT ------------
Event: time 1596217966.832653, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3642
Event: time 1596217966.832653, -------------- SYN_REPORT ------------
Event: time 1596217966.838278, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3730
Event: time 1596217966.838278, -------------- SYN_REPORT ------------
Event: time 1596217966.847659, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3817
Event: time 1596217966.847659, -------------- SYN_REPORT ------------
Event: time 1596217966.853309, type 3 (EV_ABS), code 0 (ABS_X), value 8022
Event: time 1596217966.853309, type 3 (EV_ABS), code 1 (ABS_Y), value 5955
Event: time 1596217966.853309, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3784
Event: time 1596217966.853309, -------------- SYN_REPORT ------------
Event: time 1596217966.862761, type 3 (EV_ABS), code 0 (ABS_X), value 8044
Event: time 1596217966.862761, type 3 (EV_ABS), code 1 (ABS_Y), value 5948
Event: time 1596217966.862761, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3750
Event: time 1596217966.862761, -------------- SYN_REPORT ------------
Event: time 1596217966.868258, type 3 (EV_ABS), code 0 (ABS_X), value 8068
Event: time 1596217966.868258, type 3 (EV_ABS), code 1 (ABS_Y), value 5932
Event: time 1596217966.868258, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 3749
Event: time 1596217966.868258, -------------- SYN_REPORT ------------
Event: time 1596217966.877739, type 3 (EV_ABS), code 0 (ABS_X), value 8091
Event: time 1596217966.877739, type 3 (EV_ABS), code 1 (ABS_Y), value 5912
Event: time 1596217966.877739, -------------- SYN_REPORT ------------
Event: time 1596217966.883208, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 937
Event: time 1596217966.883208, -------------- SYN_REPORT ------------
Event: time 1596217966.898210, type 4 (EV_MSC), code 4 (MSC_SCAN), value d0045
Event: time 1596217966.898210, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1596217966.898210, type 3 (EV_ABS), code 0 (ABS_X), value 8071
Event: time 1596217966.898210, type 3 (EV_ABS), code 1 (ABS_Y), value 5821
Event: time 1596217966.898210, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1596217966.898210, type 3 (EV_ABS), code 26 (ABS_TILT_X), value 1
Event: time 1596217966.898210, type 3 (EV_ABS), code 27 (ABS_TILT_Y), value 1
Event: time 1596217966.898210, type 3 (EV_ABS), code 40 (ABS_MISC), value 0
Event: time 1596217966.898210, -------------- SYN_REPORT ------------
Event: time 1596217966.907645, type 1 (EV_KEY), code 320 (BTN_TOOL_PEN), value 0
Event: time 1596217966.907645, type 3 (EV_ABS), code 0 (ABS_X), value 8053
Event: time 1596217966.907645, type 3 (EV_ABS), code 1 (ABS_Y), value 5800
Event: time 1596217966.907645, -------------- SYN_REPORT ------------
Event: time 1596217966.913218, type 3 (EV_ABS), code 0 (ABS_X), value 8038
Event: time 1596217966.913218, type 3 (EV_ABS), code 1 (ABS_Y), value 5776
Event: time 1596217966.913218, -------------- SYN_REPORT ------------
Event: time 1596217966.982200, type 1 (EV_KEY), code 321 (BTN_TOOL_RUBBER), value 0
Event: time 1596217966.982200, -------------- SYN_REPORT ------------
```
for an eraser tap. BTN_TOOL_PEN still gets set for proximity, but when it detects that it's inverted it enforces the mutual exclusion, though I think another proximity event fires when it touches because the mutual exclusion gets broken. But it seems that ChromeOS mostly cares about BTN_TOOL_RUBBER being 1 and BTN_TOOL_PEN being 0 when BTN_TOUCH happens.

sebanc, 2020-09-05:
Thanks for investigating this issue and providing a kernel patch, I will try to implement it in a future release. Do you know if other surface devices are affected or only Surface Go ?

sebanc, 2020-09-20:
@shinyquagsire23 I applied your patch in the new release

shinyquagsire23, 2020-10-02:
@sebanc sorry for the late reply, but as far as other devices being affected, I'd hazard to guess it's not actually Surface-specific as much as it's related to the Elantech panel used, and just going off of all the other libinput issues, it's likely that there's other slate/hybrid form laptops which the mutual exclusion patch would help fix.

Anyhow it seems to be working well on my Surface Go now so I'll go ahead and close the issue ^^

closed 2020-10-02 by shinyquagsire23