sebanc/brunch #387: Some Bugs  in stable release (latest)

divya789vidya, 2020-07-11:
> **None of the internal drive is mounted**

[image]

> **Andriod app loading failed**

[image]

after sometime
[image]
[image]

> **Bluetooth Problem**

[image]

DMESG TXT FILE
[dmesg2.txt](https://github.com/sebanc/brunch/files/4906115/dmesg2.txt)

_Originally posted by @divya789vidya in https://github.com/sebanc/brunch/issues/383#issuecomment-656963755_

divya789vidya, 2020-07-11:
**Audio switch Problem**
when headphone is inserted and audio output is switched to internal speaker NO SOUND came from the speaker

sebanc, 2020-07-11:
I don't mind that you have opened another issue but I will close #383.

Bluetooth problem & android problem: I am working on it, both are probably related.

mount_internal_drives: What's your setup ? Dual boot ? With which OS ? Could you provide the output of "sudo lsblk" ?

Audio switch Problem: ChromeOS seems to use specific audio interfaces names to manage audio, probably through "cras". Unfortunately, I don't know enough on linux audio to fix this kind of issues. We need a linux audio expert in the community.

divya789vidya, 2020-07-11:
Triple boot =ubuntu 20,windows10,chrome os 
file system used in the chrome os =ext4
lsblk 
[lsblk.txt](https://github.com/sebanc/brunch/files/4906708/lsblk.txt)

sebanc, 2020-07-11:
Could you try the release here ? (validate the github invite first)
https://github.com/sebanc/brunch-testing/releases

divya789vidya, 2020-07-11:
i tried the testing release  and got a bug 
[image]
the bluetooth was  showing but not working , just I turned on the buetooth in my mobile and searched no  bluetooth devices shown  in mobile as well as in the chrome os
And the most annoying problem was that when i tried to connect wifi with my mobile hotspot , it suddenly went to restart (it was just happened in the previous to previous testing release) and then after restart i once again tried to connect wifi and same again, pc restarted again .......
dmesg txt in that unstable release  
[dmesg2.txt](https://github.com/sebanc/brunch/files/4907068/dmesg2.txt)

I Just rolled back to previous stable release.....

sebanc, 2020-07-12:
Could you try the new unstable build ? (use "options=rtl8723de")

divya789vidya, 2020-07-13:
i tried the build now and got some bugs ;
wifi was not working (after enabling the rtl8723de option and removing both  tried  )
bluetooth was showing but not working (just tried to connect my phone but not showing anything else on bluetooth search ) 
dmesg txt for this build was 
[dmesg.txt](https://github.com/sebanc/brunch/files/4910943/dmesg.txt)

sebanc, 2020-07-13:
Could you try again the latest build and replace the content of /etc/init/rtl8723de.conf with:
```
start on stopped udev-trigger

script
	modprobe mac80211
	insmod /lib/modules/rtl8723de.ko
end script
```

debendrakumar, 2020-07-13:
Ok will try

divya789vidya, 2020-07-13:
i tried this one and wifi started working .
But the bluetooth problem still exist i.e. the bluetooth is not showing any device and not showing in other devies also

sebanc, 2020-07-13:
Apparently you are trying to pair your phone which is not possible in chromeos as you have to setup your phone through the dedicated menu.

Chromeos menu only pairs input/sound devices, try with that kind of device. If it does not work you are probably affected by the issue "Bluetooth working from command line but not from the settings menu" on which help is needed #331.

divya789vidya, 2020-07-13:
i didnt understand what you said in first sentence ,,,,  
and want to know that how to connect a bluetooth device by command line

divya789vidya, 2020-07-13:
no problem on the how to connect  a bluetooth device by command line .... got it

sebanc, 2020-07-13:
First sentence: you cannot pair a phone in chromeos through the bluetooth menu, use "Connected devices" menu to connect your phone.

use "bluetoothctl" to connect a bluetooth device by command line.

divya789vidya, 2020-07-15:
is there any way to install directly a driver to the chrome os  just like ubuntu
can we do by crostini

sebanc, 2020-07-15:
Not with crostini but with chromebrew it might work, what driver are you trying to install ?

divya789vidya, 2020-07-17:
i am try to install the rtlnew git driver

divya789vidya, 2020-07-17:
can we install dkms ?

sebanc, 2020-07-18:
You can build the kernel module but not install it with dkms. For now try to launch your built from source module manually in crosh shell using "insmod" command and if it works better I will help you setup a bootscript so that it runs automatically.

divya789vidya, 2020-07-19:
can you help with this one the driver rtw88 
[image]
[image]
[image]
[image]
[dmesg.txt](https://github.com/sebanc/brunch/files/4943148/dmesg.txt)
 i tried to install these driver with the help of make but unsuccessful  
https://github.com/lwfinger/rtw88.git
repository for driver

sebanc, 2020-07-19:
It's an issue with the gcc present in chromebrew, you have to add "-fstack-protector-all" to gcc arguments in the makefile.

divya789vidya, 2020-07-19:
[image]
where i have to add this line..... -fstack-protector-all

kishanrajbhat, 2020-08-20:
can I also get the latest testing brunch release? for me the bluetooth driver isnt showing up at all in settings and in terminal

khanhtranngoccva, 2021-01-08:
Well for the Android apps, I found out that if you try to sign in without having this line appear in the dmesg log, the Android container will fail to start. So I would like to have a script which checks the presence of the 32-bit capabilities line before the UI could fire up, since the container only finishes preloading after this. However, I don't know where exactly to put this into the init folder.
  while true
  do
    if [ $(dmesg | grep "32-bit capabilities" | wc -l) -gt 0 ]
    then
      break
    fi
  done