sebanc/brunch #388: Out of space errors on a 128GB USB thumb drive

fifn2, 2020-07-11:
The shell output for:
```
sudo bash chromium-install.sh -src chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin -dst  @/dev/sda
```
[image]
[image]

hbarsaiyan, 2020-07-11:
> The shell output for:
> 
> ```
> sudo bash chromium-install.sh -src chromeos_13020.87.0_rammus_recovery_stable-channel_mp-v2.bin -dst  /dev/sda
> ```
> 
> [image]
> [image]
> 
> It doesn't boot.

Format /dev/sda and do a reboot. After that it should work fine.

Skawtnyc, 2020-07-11:
Make sure to clean off all of the existing partitions on the USB stick. If you have a Windows system, diskpart is perfect for this.

sebanc, 2020-07-13:
If I am not mistaken, this issue was due to an old cgpt version that came with ubuntu 16.04 and has been fixed.

closed 2020-07-13 by sebanc