sebanc/brunch #389: Usage syntax error

ranjith6303, 2020-07-11:
Chromeos-install. sh: line 7: syntax error near unexpected token `$’\r’ ‘
‘hromeos-install. sh: line 7: `usage( )

How can i fix this

ranjith6303, 2020-07-11:
[image]

sebanc, 2020-07-13:
As mentioned in "Please read this before creating issues #307", I am not related to any youtube video and don't provide support if you have installation issues while following one.

If you have issues with the official way in the readme, I will try to help you. Else please submit your issue to Kedar.

closed 2020-07-13 by sebanc

ranjith6303, 2020-07-14:
Sir i am unable to  run this code
sudo bash chromeos-install.sh -src < path to the ChromeOS recovery image > -
dst chromeos.img
please give me a solution

*Yours faithfully,*

*RanJith Tr,9400016777,8101555000*