sebanc/brunch #390: mount_internal_drives option: mount path changed

jhedfors, 2020-07-12:
@sebanc Could the mountpoint be changed to match the partition name?  It seems llike the name that are given are perhaps dependant on the order in which they are read, which is not consistant. I have shortcuts that I have made that are broken (likely after a Brunch rebuild).

My current mapping:
sda                              8:0    0 167.7G  0 disk 
|-sda1                           8:1    0   600M  0 part 
|-sda2                           8:2    0  54.8G  0 part /media/removable/External Drive 1
|-sda3                           8:3    0  63.5G  0 part /media/removable/External Drive 2
`-sda5                           8:5    0  48.8G  0 part /media/removable/External Drive

Could the (not "External", nor "removable" :) haha...) mountpoints instead be something like this?:
- Drive SDA2
- Drive SDA3
- Drive SDA5

sebanc, 2020-07-13:
Sorry but all that is managed automatically by cros-disks which does not allow much customization, I don't think I can do much about that.

jhedfors, 2020-07-13:
I did not realise that you implemented an existing tool. That is totally fine, and I very much appreciate having the option available. I just don't use the shortcut function.  Thanks!

closed 2020-07-13 by jhedfors

jhedfors, 2020-08-01:
@sebanc 

FYI/SOLUTION: I discovered that cros-disks uses the disk label if it exists!