sebanc/brunch #391: Question: Can you update Brunch externally?

jhedfors, 2020-07-12:
Is there a method to upgrade Brunch externally (live USB)?  I want to put this on a family members computer with auto-updates on. If by chance it doesn't boot (due to old Brunch), can it be upgrades short of re-installing and starting over?

Releted question - it there a way to get an alert when a new Brunch is available, or any automated way to update?

sebanc, 2020-07-12:
To update brunch from linux, extract rootc.img from the brunch release and use "dd" or "pv" to write it to partition 7. (If you use dual boot, you will need to use "losetup" first to create a loop device and then write to the partition 7 of the loop device.

To notify about brunch updates, I would probably have to write a chrome extension but you can probably have the same result by using an already existing extension such as "distill-web-monitor" to monitor the brunch release page.

jhedfors, 2020-07-13:
Thanks for the suggestion, hopefully we will stay on top of it, and it won't come to that.  I have set up IFTTT to monitor the release page (https://github.com/sebanc/brunch/releases.atom) -  It looks like to may be doing the same thing as "distill-web-monitor". I will give that a try if IFTTT fails me. :)

closed 2020-07-13 by jhedfors

jhedfors, 2020-07-14:
I created this script to automatically download and install once I know there is an update:

latest=$(curl -s https://api.github.com/repos/sebanc/brunch/releases/latest | grep -o "http.*tar.gz"); curl -L $latest -o ~/Downloads/latestbrunch.tar.gz; sudo chromeos-update -f ~/Downloads/latestbrunch.tar.gz