sebanc/brunch #392: Camera not detected - Surface Pro 6

BuzzKnightbeer, 2020-07-12:
Hello,

I just prepared a USB stick and everything except the Camera seems to be working great.

Before creating a dual boot setup, I just wanted to see if this is fixable.

I used the latest Brunch (Brunch r83 k4.19 stable 20200708) and the 83 rammus recovery image.

I've read in a different post that using samus might fix the camera issue but before testing it I wanted to make sure that this won't introduce new problems as in the readme it says ramus is only for 4th generation and older CPUs. The Surface Pro 6 has an 8th generation CPU.

Does anyone have any experience with this?

Also, unrelated to this, what is the difference between Brunch and Crostini? I haven't used the later because Brunch works nearly perfect so far. Just curious...

Skawtnyc, 2020-07-12:
This is a Linux driver issue. As in, there are no Linux drivers for Surface cameras. My solution was to simply get an inexpensive external Logitech webcam, which works fine.

BuzzKnightbeer, 2020-07-12:
Ahh sorry, I meant Crossaint. I edited the original post.

Is there a way to add support for the cameras manually?

Skawtnyc, 2020-07-12:
None that anyone is aware of. It seems that Microsoft hasn't shared any code or specs for the hardware, so no drivers are available except for Windows drivers.

sebanc, 2020-07-13:
Indeed I confirm what @Skawtnyc says, there is currently no linux kernel driver for Surface devices camera (still the nice guys at linux-surface github seem to be working on it) and "samus" recovery image will not help without driver, you should therefore stay on "rammus".

With regards to the difference between croissant and brunch, you can find a quick summary in the issue #59 .

BuzzKnightbeer, 2020-07-14:
Thanks, got it!

closed 2020-07-14 by BuzzKnightbeer