sebanc/brunch #393: Toggle WIFI after sleep to turn on to get WIFI working

tienhuynh5312, 2020-07-13:
Laptop spec:
Thinkpad T480 i5 8250u
ChromeOS Image Recovery: nami

My wifi card is BCM4360 [14e4:43a0]
Whenever after wake from idle/sleep, i need to toggle the wifi button on/off to get the wifi connected again.

sebanc, 2020-07-13:
Could you try the latest release available below ? (after validating the github invite email)
https://github.com/sebanc/brunch-testing/releases

tienhuynh5312, 2020-07-13:
is options=broadcom_sta available too? broadcom_wl performance is really good anyway, but I want to check it out just for comparison.

sebanc, 2020-07-13:
Forgot to answer that part, "broadcom_wl" and "broadcom_sta" are actually the same, I just did not know which name to give it and I think someone asked for it to be renamed but it is the same driver.

tienhuynh5312, 2020-07-13:
@sebanc just checked it out. The issue does not go away.

sebanc, 2020-07-13:
Ok I will try to see if I can find something else

sebanc, 2020-07-13:
Have you tried to turn off "Keep Wifi on during sleep" privacy and security setting ?

tienhuynh5312, 2020-07-13:
Did it, and unfortunately the issue still exists

sebanc, 2020-07-13:
Could you attach the output of "lsmod" to this issue ?

tienhuynh5312, 2020-07-13:
here you go
[lsmod.txt](https://github.com/sebanc/brunch/files/4914030/lsmod.txt)

sebanc, 2020-07-13:
I am not sure if opencore would change something, could you try adding "module_blacklist=thinkpad_acpi" to the kernel command line ? (not sure it is related but I find it weird that a thinkpad module is loaded)

sebanc, 2020-07-13:
Also you can try running after boot:
`iw dev wlan0 set power_save off`

and see if wifi stays on after suspend

tienhuynh5312, 2020-07-13:
How can I execute the command? chrome shell said no command found.

sebanc, 2020-07-13:
Sorry forgot to start the command with "sudo"

tienhuynh5312, 2020-07-13:
The issue still persists. I guess I have to live by this. This is not a chromebook so I would not expect it to be perfect. So far 99% things are working really good. Just this issue to make this T480 "chromebook" perfect :).

tienhuynh5312, 2020-07-14:
@sebanc 
```
[16778.668284] PM: suspend exit
[16778.739994] usb 1-7.3: new full-speed USB device number 14 using xhci_hcd
[16778.819658] usb 1-7.3: New USB device found, idVendor=05ac, idProduct=828d, bcdDevice= 1.50
[16778.819667] usb 1-7.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16778.819673] usb 1-7.3: Product: Bluetooth USB Host Controller
[16778.819677] usb 1-7.3: Manufacturer: Apple Inc.
[16778.819862] usb 1-7.3: Device is not authorized for usage
[16778.826401] usb 1-7.3: authorized to connect
[16778.932506] Bluetooth: btbcm.c:btbcm_setup_apple() hci0: BCM: chip id 73 build 1125
[16778.933500] Bluetooth: btbcm.c:btbcm_setup_apple() hci0: BCM: product 05ac:828d
[16778.934505] Bluetooth: btbcm.c:btbcm_setup_apple() hci0: BCM: features 0x07
[16778.950507] Bluetooth: btbcm.c:btbcm_setup_apple() hci0: BCM20702B0 Generic USB Class 1 @ 20 MHz
```
after suspend exit, I did not see any wakeup call for wifi. Was that an issue?

sebanc, 2020-07-14:
It seems normal as far as I can tell, the problem is most likely with the network manager (most linux distro disable it before sleep and re-enable it after sleep).

If you run "udevadm monitor" in crosh shell and disable/re-enable wifi in the settings panel. Is there some output in crosh shell ?

tienhuynh5312, 2020-07-14:
@sebanc there is no output when I did that. But before that there is some other stuff.
[image]

sebanc, 2020-07-14:
Ok, that's too bad...

Could you try to run "echo on | sudo tee /sys/bus/pci/devices/0000\:03\:00.0/power/control" in crosh shell before suspend ?

tienhuynh5312, 2020-07-14:
it is `on` before and after the suspend

sebanc, 2020-07-14:
ah ok, I will try to think about something else

tienhuynh5312, 2020-07-15:
@sebanc  in broadcom_wl patch script, can you add
```
blacklist b43legacy
blacklist b44
blacklist wl
```
I remembered i need to add these and other blacklist lines in your broadcom_wl to get the wifi working as other guides suggested.

sebanc, 2020-07-15:
It's already added in /etc/modprobe.d/broadcom_wl.conf

sebanc, 2020-07-15:
Could you try with "options=android_init_fix" ?

tienhuynh5312, 2020-07-15:
Still no luck. I tried to use other patched broadcom_wl drivers on github, but there were some errors about linux headers. Still figuring out. Should newer kernel help at all?

sebanc, 2020-07-18:
It would probably not help, I have another similar report #'403 that you have commented for "iwlwifi_backport" option which works in a similar way as "broadcom_wl". I wonder if it is not related to a current chromeos bug.

Could you try beta or dev channel ?

tienhuynh5312, 2020-07-18:
Beta still no luck. This is a familiar issue in the hackintosh world as I am hackintoshing right now. Let's me do some research on this from hackintosh side, so it could shed some light on this