sebanc/brunch #395: Sleep instead of power menu with acpi_power_button?

hacker1024, 2020-07-13:
Hi, I have to use `acpi_power_button` for power button functionality on my tablet (system info described in #394). As it's a tablet, there's no easy way to sleep it - the power button brings up a power menu (which does not have a sleep option), and the only other ways to sleep the tablet are with a keyboard shortcut (I have no keyboard) and letting it time out.

I don't know how your power button patches work, so I'll understand if it's too hard to do this, but would it be possible to map the power button to trigger sleeping instead of the power menu?

sebanc, 2020-07-13:
Could you leave the option "acpi_power_button", remove "--aura-legacy-power-button" and see if it works as expected ?

hacker1024, 2020-07-13:
Thanks, this fixes it (although there's no animation, but that's minor detail). I also added `--force-tablet-power-button` to enable the button while in laptop mode.

I've noticed another issue while doing this, though: #398.

hacker1024, 2020-07-13:
Oh, I forgot - this seems to break the hold-down menu (which is to be expected as it seems the button does not send a release event). That menu has no unique functionality, though, so it's not really an issue.

sebanc, 2020-07-13:
As you mention this is expected as the button does not send a release event.

closed 2020-07-16 by hacker1024