sebanc/brunch #397: Brunch crashes occasionaly on Inspiron 3583

ilovemc888, 2020-07-13:
I am using a triple-boot setup. Since my Windows and KUbuntu installs are encrypted (Bitlocker and LVM), I had created separate partition (64G) for Brunch. Usually, it works pretty well, but I have been having issues that sometimes it will start crashing, causing SIGBUS(8) error on any website visit, and after about 30sec, it will go to console and have lots of errors on loop0p3, and remount root readonly. It will later kernel panic and reboot. When my laptop reboots, apparently (idk why) it goes to memory test, will beep with no errors, and will keep doing that until i turn my laptop off, and turn it back on. How do I fix this. Chrome OS v83, Brunch v83. For now I am using Sarien board, as it is most compatible with my CPU (i7-8565U), however i tried samus and eve and have same problem. Please help me fix.

ilovemc888, 2020-07-13:
[image]
looks like this after crash

sebanc, 2020-07-13:
Hi,

It looks like an error with the partition the chromeos.img file is on. How did you format it ? (ntfs or ext4)

ilovemc888, 2020-07-13:
I used ext4. It works sometimes, and works pretty well when it does work.

sebanc, 2020-07-13:
Unfortunately, I have never encountered that issue and I also have no other reports.

From what I saw by looking at this kind of issues, it could be caused by:
- Windows fast startup.
- An invalid BIOS time/date (in the future).

Also your setup (Bitlocker+LVM) is quite unusual, you might try the option "mount_internal_drives" (it will not allow mounting of your drives due to their format but it might prevent crashes).