sebanc/brunch #399: Ext 4 or ntfs

CornishRattler, 2020-07-13:
Not an issue so delete if not allowed, I currently have W10 and Chrome OS as a dual boot with Chrome being on an ntfs partition on the HDD. My question is would changing this partition to Ext4 have any benefits?

sebanc, 2020-07-13:
Even though it seems pretty reliable currently, the ntfs partitions management by the linux kernel is not great so I would recommend ext4.
Aside from that, according to benchmarks performed by users, the performance is the same on ext4 or ntfs.

NSbuilder, 2020-07-16:
> Even though it seems pretty reliable currently, the ntfs partitions management by the linux kernel is not great so I would recommend ext4.
> Aside from that, according to benchmarks performed by users, the performance is the same on ext4 or ntfs.

I don't understand... doesn't brunch choose the filesystem automatically? I deleted all partitions before installing brunch.

forumi0721, 2024-03-10:
> Even though it seems pretty reliable currently, the ntfs partitions management by the linux kernel is not great so I would recommend ext4. Aside from that, according to benchmarks performed by users, the performance is the same on ext4 or ntfs.

Hi, I have a question.
I thinks ntfs3 is enough good. (Not ntfs-3g)
What is a driver use in Brunch?  traditional ntfs? or ntfs-3g? or ntfs3?