sebanc/brunch #400: Chrome OS boots only from usb's bootloader after installing on HDD

desmondam, 2020-07-13:


sebanc, 2020-07-13:
Regarding boot, there is a bug with ubuntu's grub binary. Replace your ubuntu grubx64.efi (usually located in /boot/efi/efi/ubuntu) by the brunch one that you can find on partition 12 or on github branch r83 in "efi-mods/boot/".

Regarding lag, not sure what you mean, people tend to think chromeos is incredibly fast and has perfectly fluid animations but that's not actually the case. You can probably find online resources on how to disable animations.

tornadox, 2020-07-17:
@desmondam I think you are using an underpowered device that barely can work any modern OS with a graphical UI. I tried Chrome OS on z3735 - it works, the question is what your are plannig to do on it, it runs android apps, the original android 4.4 also managed to run android apps, but that's purely limited to multimedia. And these stripped CPUs are 6 years already old now.

I have an original chromebook with i3-8130u it's really fast, I tried brunch on a gaming laptop Razer Blade - it's also very fast. There should a self-understanding that there is no modern commercial OS that tends to work great on old budget hardware.