sebanc/brunch #402: brunch USB fails to boot with illegal instruction

da0ist, 2020-07-14:
ThinkPad X220 brunch 83 USB stick created on ElementaryOS.
[image]

sebanc, 2020-07-14:
It's 2nd gen, try "samus" recovery image.

da0ist, 2020-07-14:
I could have sworn I used rammus before say around 80? Will try sammus, thanks!

sebanc, 2020-07-14:
You are right google changed something to "rammus" in r83 which makes it incompatible with Intel Gen < 4

da0ist, 2020-07-14:
Looks like it's working, but the screen went to sleep and I can't wake it. I see disk activity (during the install) so I'll just have to trust the process. It'll shut down if it completes successfully, right?

sebanc, 2020-07-14:
No it does not shutdown at the end, you should probably do it again

da0ist, 2020-07-14:
It eventually woke up and installed fine! I am posting this from it. My ThinkPad X201 won't boot from the stick, is it too old (lacks secure boot?)

sebanc, 2020-07-14:
It does probably not support uefi boot, you can try the mbr_support.tar.gz package in the master branch of this repo (extract the brunch release, then the mbr_support.tar.gz in the same folder and install as usual)

da0ist, 2020-07-14:
I'll give that a try after I Rammus my ThinkPad X1C6! Thanks for this awesome project!

closed 2020-07-14 by da0ist