sebanc/brunch #403: WiFi Intel 9560 issues with Android subsystem

CongoKong, 2020-07-14:
Specs:
- intel i5-10210U laptop
- rammus recovery image
- intel 9560 wifi

The intel 9560 wifi card is not supported out of the box. So I enable it using the iwlwifi_backport option. This works quite well except in two situations:
1. Waking from sleep. The card cannot connect to the wifi network again. This worked with "Brunch r83 k4.19 testing 20200704" but does not work with "Brunch r83 k4.19 stable 20200708".
2. Downloading apps from Google Play Store in the Android subsystem. This never worked. After clicking "install" the download starts but never progresses. It is stalled at Downloading: 0%. I need to connect a phone and enable USB tethering to make Play Store downloads work.

In general connecting to (Google?) servers can take quite some time in the Android subsystem while working flawlessly in ChromeOS. No problem with USB tethering though in the Android subsystem so I guess maybe there are some issues with the i9560 driver?

tienhuynh5312, 2020-07-18:
Did you try other images? Choose an image that have the same CPU and same manufacturer so that your device is likely compatible. Even though sebanc recommends to choose Rammus, but my thinkpad t480 is very good with Nami which is for 8 gen cpu.

CongoKong, 2020-07-18:
I did try other images: nami, hatch... they work but without native WiFi support. I think I got somewhat unlucky with the i9560. 8th gen chromebooks use i7xxx cards, 10th gen use AX200. I do not know of a 9th gen chromebook image. I think I need to change my WiFi card.

CongoKong, 2020-08-16:
added AX200 WiFi card - no change. Card not recognized. ChromeOS seems to be *very* picky when it comes to WiFi. The AX200 card works perfectly in Linux and Windows - even without changing drivers... I wonder which WiFi 6 card is used in "hatch" devices like Samsung Galaxy Chromebook or ASUS Chromebook Flip C436FA. Is it AX201?

sebanc, 2020-08-29:
@CongoKong Added you to "brunch-next" which has 5.4 kernel, could you try to see if it works better ?

CongoKong, 2020-08-30:
@sebanc tried Intel AX200 WiFi card with hatch recovery image / Brunch r83 k5.4 unstable 20200711 and... ta-dah! Success! WiFi card is recognized in ChromeOS 84.0.4147.136 and works like a charm. iwlwifi_backport option not required anymore. (Actually enabling iwlwifi_backport for this brunch-next version resulted in quite unstable network connectivity).

Bluetooth also works nicely.

Google Play Store stuff is now downloading like on a real Android phone and as an added bonus the "pauses for thought" where the Android subsystem seemed to cause general 20 - 30 second freezes in ChromeOS are gone! (Need to check this more thoroughly though.)

Generally very pleased with this version of the framework and it is not that unstable (at least not for me).

closed 2020-09-05 by CongoKong

matthiasjo, 2020-10-26:
@sebanc I have the exact same issue. I have a Dell XPS 9560 with Intel AX200 wifi. I enabled iwlwifi_backbort which allowed me to scan for wifi networks but I can't connect to any. Could I also try the "brunch-next" 5.4 kernel? And thank you for this awesome project.

sebanc, 2020-10-26:
@matthiasjo The github invite has been sent

matthiasjo, 2020-10-26:
> @matthiasjo The github invite has been sent

thx. new kernel fixed it.