sebanc/brunch #406: Chuwi Hi10 x Detachable Tablet compatibility report. (Works except camera and stylus)

Ebbot9876, 2020-07-16:
I now have had this tablet for over a month, and it´s working fine with Brunch r83/Rammus!
All Android apps i tried is working.
Linux is there and works.
Bluetooth works.
Wifi works.
Camera works but picture on screen is rotated and mirrored. (I´ve mentioned this in #276)
Microphone works.
Touchscreen works, but supplied stylus does not!
Trackpad and gestures works.

I really appreciate the work you are doing here!
[lsmod_HI10X.txt](https://github.com/sebanc/brunch/files/4932500/lsmod_HI10X.txt)
[dmesgHI10X.txt](https://github.com/sebanc/brunch/files/4932501/dmesgHI10X.txt)

Ebbot9876, 2020-07-20:
@sebanc is Therese anyway to enable stylus and turn camera right?

closed 2020-07-28 by Ebbot9876

CWayman, 2020-07-29:
You closed this ticket, is it fixed?

Ebbot9876, 2020-07-29:
I´ve opened two separate issues #431 and #432 instead!

ccchan234, 2023-02-28:
fydeos v15.1 has these 2 things worked, in 2023.

i'll guess chromeos or os flex also support it in 2023, others could try.