sebanc/brunch #407: Lenovo ThinkpadT520 Touchpad issue

fruchttiger87, 2020-07-16:
Hi sebanc,
Hi everybody,
I tried to set up my old Lenovo Thinkpad T520 with Chrome OS. The touchpad is detected und the mouse works. But it the cursor reacts completly different compared to Linux, crouton and Windows. Even in crouton it moves smoothly and exactly. In Chrome OS the mouse cursor jumps sometimes when I leave the finger from the touchpad. Additionally the mouse cursor does not react that exactly as it should. (I hope you can imagine what I try to describe. It is not easy for me because english is not my native language).

I red what you wrote about touchpad issues. Before I try hundred of configs I wanted to ask If somebody knows the problems and solved it. Perhaps the configs won't change something because the issue has its root somewhere else? 

I hope you can help me. I love your project!!!!!

loogol, 2020-07-21:
i can confirm this issue on the Lenovo v580c. and my touchpad VERY sensible - its reacts to finger in 1-2 mm of touchpad

jafer241177, 2020-08-09:
same problem with lenovo ideal pad flex 3 11 GL05

closed 2026-04-20 by github-actions[bot]