sebanc/brunch #408: Chrome Cpu Consumption?

roynatech, 2020-07-17:
I have installed altas image with the latest brunch framework. It works well including wifi gpu sound etc. One problem is cpu load. Process named 'Browser' (Probably chrome) takes about 20~30% of cpu. And the pc was idle at the moment. Is chrome os heavy like that? I think it can be hardware issue 'cause my cpu is Intel Core i5-5200U (2core 4thread 2.2Ghz, boost clock 2.5Ghz) 
I wanted a different OS than Windows and Linux, so I installed this. But now, the fan sound disturbs me... It's too loud even though the computer being idle.

roynatech, 2020-07-17:
[image]
Attaching screenshot

Crescendo-BLYAT, 2020-07-17:
Browser itself is the OS & 30% is not heavy....
maybe you could clean the heatsink & change the paste if it's begin heating up....

I could run some Android apps + some tabs on my Celeron N3350 & the CPU usage only jumped to 32%...

so it's not heavy, unlike Windows 10 which can go to constant 70% idling in the very same laptop....

tornadox, 2020-07-17:
Have you tried other recoveries?

sebanc, 2020-07-18:
Atlas has an ipu3 camera framework so the android camera service fails and runs in a loop in the background. You should probably use "rammus" or "eve" instead (refer to issue #359).

rayqusa, 2020-08-05:
whats the name of the apps that show like task manager?

closed 2020-08-29 by sebanc