sebanc/brunch #409: how to mount a HDD partitions (ext4 or NTFS) to chrome OS which can also be viewed on both dual booted windows

srikarsashankbandi, 2020-07-17:
I tried an optional step from readme by editing grub config with " mount_internal_drives" also did a powerwash after editing but couldn't mount. Any suggestions?

sebanc, 2020-07-18:
Could you attach a dmesg to this issue ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your downloads folder)

srikarsashankbandi, 2020-07-18:
Hi sebanc, many thanks for your brunch kernel. here is your requested dmesg.
[dmesg.txt](https://github.com/sebanc/brunch/files/4942926/dmesg.txt)

sebanc, 2020-07-19:
You don't have the "options=mount_internal_drives" in your kernel command line, as you are using dual boot you have to add it to the linux grub configuration you have created.

srikarsashankbandi, 2020-07-19:
I did add the options=mount_internel_drives in the kernel command line.
here is the screenshot and dmesg after.
[image]
[dmesg.txt](https://github.com/sebanc/brunch/files/4943283/dmesg.txt)

sebanc, 2020-07-19:
No still not:
```
[    0.000000] Command line: BOOT_IMAGE=(loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug loop.max_part=16 img_part=/dev/sda10 img_path=/chromos.img
```

You are modifying the wrong grub config, in dual boot you have to modify the linux one that you created at install time.

srikarsashankbandi, 2020-07-20:
Hi sebanc, could you please elaborate on the steps where I have to change to get it. if that is on the grub menu entry, here is the screenshot.
[image]

djgerman10, 2020-07-20:
https://www.youtube.com/watch?v=bJwVL4QjYUw

ajudi46, 2020-07-30:
Hey @srikarsashankbandi did you got the answer as to where to put the 'options=' in the grub file . I also dial booted with grub2win

mhdrasel, 2020-08-04:
> Hi sebanc, could you please elaborate on the steps where I have to change to get it. if that is on the grub menu entry, here is the screenshot.
> [image]

Just after cros_debug write
 options=mount_internel_drives
That's all. Good luck

derqami, 2020-11-09:
Hi,
 Is it possible to mount a specific partition on booting? So that the other system partitions (linux, windows) would not be in danger of any mistake. Thank you.

ameyjoshi3068, 2020-11-09:
@derqami Hi,
Yes this is possible to mount a specific partition  in crosh. For that
1.Identify your My Files path.
    There is a trick to to find it: create a text file in text app, save it to my files and open the file in chrome browser(Right click and choose `view`) in the address bar there is path for your text file. copy it such that it is in format `/home/................................../MyFiles`
I cant remember whether last word is myfiles, my_files or My_files but it should be linke that.
2. Create directory in my files
    `mkdir -p **your_path**/partition
3.Run `sudo mount /dev/xxxx **your_path**/partition

Now you can open files app and browse those files.

**Note**: Unmount  the partition after use to be safe  `sudo umount  /dev/xxxx`

derqami, 2020-11-09:
@ameyjoshi3068 
It works, thank you. Is it possible to make another magic so the partition would be auto-mounted when chromeos boots. Or make exceptions in "mount_internal_drives" to exclude unwanted partition mount.

derqami, 2020-11-14:
Mounting a partition under 'myfiles" restricts hdd to be used by android apps.
 - download a file explorer from play store, create a new folder under "internal memory'. in my case its STRG . You can make a txt file in this folder and open with chrome browser to find out the path, just like explained above. in crosh
-shell
-"sudo mount /dev/partitionx run/arc/sdcard/write/emulated/0/STRG"
Now you can access and modify your files both in android and chromeos apps.

Zarvis12, 2020-12-10:
> @derqami
> Yes, it is possible to mount a specific partition in crosh. For that
> 1.Identify your 'My Files 'path.
> There is a trick to find it: first, create a text file in the text app, then save it to my files and open the file in the Chrome browser(Right-click and choose `view`) then in the address bar, there is a path for your text file. copy it such that it is in the format `/home/chronos/................................../MyFiles`
> 2. Create a directory in my files
> `mkdir -p **your_path**/partition`
> 3.Run `sudo mount /dev/XXXX **your_path**/partition`
> 
> Now you can open the Files app and browse those files.
> 
> **Note**: Unmount the partition after use. `sudo umount **your_path**/partition`

Can you help me to mount other partitions as removable

Zarvis12, 2020-12-10:
> Mounting a partition under 'myfiles" restricts hdd to be used by android apps.
>
> * download a file explorer from play store, create a new folder under "internal memory'. in my case its STRG . You can [...]

[image]

I cant access them in Android

closed 2026-04-20 by github-actions[bot]