sebanc/brunch #410: How to make items larger by kepping at max resolution

dubagunta-rohit, 2020-07-17:
I have been using Chrome os for the last month with dual boot. I noticed the display looked better when using Windows..
So when i was going through display settings , i found i opted for 120% in display size...i made my  resolution 1600x900..
But when i opt for 100%  the resolution improved to 1920x1080 but the text and icons are so tiny , which strain my eyes. 
**Any solution to make them biggeer without decreasing the resolution??**
[image]

sebanc, 2020-07-18:
I only see 2 options if the default chromeos scaling does not render well:
- try another scaling setting from 120% to 130% for example... but I guess you have already tried that.
- use "video=" kernel command line parameter to use a lower default screen resolution (e.g. "video=DVI-I-1:1024x768@85")

dubagunta-rohit, 2020-07-18:
@sebanc  Scaling it to 120% makes icon and text look good but screen resolution 1600x900 , which isn't sharp enough for my liking.

About the 2nd thing.... I don't want to lower my screen resolution.... I want it at 1920x1080 with larger icons and display

sebanc, 2020-07-18:
I cannot rewrite chromeos scaling code, in that case you can just hope google improves scaling at some point

dubagunta-rohit, 2020-07-18:
I was watching this video https://youtu.be/m5ShY3hiKaM
Even the Pixel book go is scaled at 1536x864 by default even though it has a 1920x1080 .

I guess you are right.. Google should improve this.
@sebanc 
[image]

tienhuynh5312, 2020-07-18:
@dubagunta-rohit after you spend some money on their products :)

dubagunta-rohit, 2020-07-18:
@tienhuynh5312  I am using a Google pixel 3. 
I definitely buy a chromebook if it outperforms windows in most aspects.. Please keep your insane suggestion to yourself.

tienhuynh5312, 2020-07-18:
@dubagunta-rohit easy buddy. What i am saying is they support the official product line not these kind of project because you need it. Now you want to kick people off because of their own opinion. You are the one should be kicked off.

dubagunta-rohit, 2020-07-18:
Scaling is something integral to an os..I'm not asking them to support any unofficial devices... 
If you have read my above comment you would have understood. The same problem is there with Pixel book go

tienhuynh5312, 2020-07-18:
@dubagunta-rohit have you tried linux scaling? They are even worse. People are complaining for years for the better support. And my comment was kind of a joke. You are just over your head.
@sebanc has this project because he wants people can experience chromeos. As he said, he cannot deal with your issue because that is Google job, not his. Aka you are barking up The wrong tree.

sebanc, 2020-07-18:
Let's calm down guys, it is just a small misunderstanding.

dubagunta-rohit, 2020-07-18:
@tienhuynh5312  bro do you even understand.... I was hoping for Google for improve it.. i was not barking at @sebanc

tienhuynh5312, 2020-07-18:
Okay I apologize you. I know you want Google to improve it. Added features are always better. I want it too.
Here is what my joke means if you and others spend tons of money on Google and all request that features, you will get that features sooner. Otherwise, it will be low priority feature and who knows when it will be added.