sebanc/brunch #413: Sign into your Chromebook

dubagunta-rohit, 2020-07-20:
This happened for the 4th time now within a span of a week. It's asking me to sign after booting up. 
Once I sign in its back to normal, no data lost. 
I'm on the lastest brunch on stable version 83.0.4103.119.

[image]

Crescendo-BLYAT, 2020-07-23:
This only can happen IF you changed your password.
Otherwise check your Google account security logs.

dubagunta-rohit, 2020-07-23:
[image]
i didnt change the password.

Crescendo-BLYAT, 2020-07-24:
Happened on my ASUS Chromebook C214MA today, so this problem is Google-side.
Not related to this firmware as my Chromebook running official firmware, not brunch.

Also my Acer Brunchbook with latest brunch + coral recovery never experienced this problem.

dubagunta-rohit, 2020-07-24:
okay cool. :)