sebanc/brunch #415: System UI is not Responding every time

efbe2121, 2020-07-20:
I've installed Chrome OS in my friend's laptop with Intel Celeron N4000, and 2GB RAM using samus image. But everytime it boots, a pop up appears that System UI is not responding, and it need a long time to close it. What if i change it to rammus image? or is it because the laptop doesnt support chrome os at all?

Skawtnyc, 2020-07-20:
The Celeron N4000 is an evolution of Apollo Lake, 7th generation Intel. You should be able to use the Rammus image. Try that and see if it fixes the problem.

efbe2121, 2020-07-21:
It doesn't fix the problem, yet it gives anoter issue that Playstore unable to load (its not the spinning circle)

Any help?

Crescendo-BLYAT, 2020-07-21:
try Octopus as it uses N4000 also

efbe2121, 2020-07-21:
As far as i concern, based on what sebanc said

**First, there is no reason to think my device is similar to this Chromebook so I should use this recovery image. Globally ChromeOS recovery images are all very similar aside from the kernel, modules, firmwares and userspace config files. All those elements are replaced by brunch when rebuilding the rootfs, so there is no reason to think that way.**

This make me thinking changing image wont do much work

sebanc, 2020-07-28:
Hi, could you try deleting google sync ?
https://support.google.com/chromebook/answer/2914794?hl=en

efbe2121, 2020-07-28:
It fixes the problem! I will inform again if this issue occurs at any point

closed 2020-07-28 by efbe2121