sebanc/brunch #416: Is "qualcomm atheros qca61x4 wireless network adapter" driver available?

cheng80, 2020-07-20:
Is "qualcomm atheros qca61x4 wireless network adapter" driver available? Driver does not run on Samsung Galaxy Book (SM-W720) model.

sebanc, 2020-07-23:
The driver is included but this laptop seems to require a specific firmware, could you unzip the below file and copy it to /lib/firmware/ath10k/QCA6174/hw3.0:
[board.zip](https://github.com/sebanc/brunch/files/4964284/board.zip)

stvrly, 2020-08-03:
Hi... I have the same laptop. Just now I tried the board.bin suggested here and it isn't working. The wi-fi still won't enable.

Here is the output of `dmesg|grep ath`:
```
ath10k_pci 0000:01:00.0: enabling device (0000 -> 0002)
ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
ath10k_pci 0000:01:00.0: qca6174 hw3.2 target 0x05030000 chip_id 0x00340aff sub 144d:c14f
ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 1
ath10k_pci 0000:01:00.0: firmware ver WLAN.RM.4.4.1-00140-QCARMSWPZ-1 api 6 features wowlan,ignore-otp,mfp crc32 29eb8ca1
ath10k_pci 0000:01:00.0: board_file api 2 bmi_id N/A crc32 4ac0889b
ath10k_pci 0000:01:00.0: htt-ver 3.60 wmi-op 4 htt-op 3 cal otp max-sta 32 raw 0 hwcrypto 1
ath: EEPROM regdomain: 0x5f
ath: EEPROM indicates we should expect a direct regpair map
ath: invalid regulatory domain/country code 0x5f
ath: Invalid EEPROM contents
ath10k_pci 0000:01:00.0: failed to initialise regulatory: -22
ath10k_pci 0000:01:00.0: could not register to mac80211 (-22)
```
Anything else I can try?

stvrly, 2020-08-03:
These seem relevant.

https://bugzilla.redhat.com/show_bug.cgi?id=1615436
https://bugzilla.kernel.org/show_bug.cgi?id=206567
https://forums.linuxmint.com/viewtopic.php?t=295347

sebanc, 2020-08-29:
@stvrly Thanks for the pointers :) I will try something in next release

stvrly, 2020-08-31:
@sebanc Well, last week I bought an actual Chromebook :) ... so I don't need this driver anymore. But maybe @cheng80 still does.

sebanc, 2020-09-05:
this patch has been included in the new r85 build

moly, 2021-09-30:
@cheng80 @stvrly hi, did either of you end up getting chromeos to run well on your galaxy book? I've just installed this on mine and I'm having problems with the screen being rotated and the touchscreen inputs being inverted that I can't resolve. You're the only people I've seen online using the same device.

moly, 2021-10-01:
Ok, thanks anyway.

@sebanc regarding the WiFi issue, I installed brunch r91 and still had the same error. So I deleted board.bin and board-2.bin from /lib/firmware/ath10k/QCA6174/hw3.0/, then copied in the board.bin from your post above, rebooted and now WiFi is working. Thanks