sebanc/brunch #418: Chrome OS dual boot

CornishRattler, 2020-07-21:
Weird one started today, been using Chrome OS (rammus) dual boot with W10 on a Dell latitude 5290 2in1 happily for a while. Today I can only boot into Chrome OS after booting W10 and restarting the machine. if i choose to boot straight into Chrome I just get this........."your system is repairing itself please wait" and a 5 min countdown. Let that go through and it starts again with the same repair message.

Boot into W10, restart and all is fine. Any clues?

kar200, 2020-07-22:
I had the same happen to me yesterday and just reinstalled (I am new to the whole Chrome OS thing). I though I might have done something wrong.

Using Rammus and latest stable on a Huawei Matebook 13 (2020) in triple boot (Windows 10, Ubuntu 20.04 and Chrome OS)

dubagunta-rohit, 2020-07-22:
Its just because of windows fast startup...
Solution
1.)boot into windows , search for power options on windows search >then click on  Edit power plan > then click on power options > then click on "choose what power button does"  > click on change settings that are currently unavailable > ****Untick TURN ON FAST  STARTUP **** and save changes.

2.) Everytime you Shutdown windows, you need   ****hold  on the Shift key ****   then Start > Power > Shutdown.

efbe2121, 2020-07-24:
Check if your windows 10 is hibernating, for some reason, i cant boot into Chrome OS with same message "Your system is reparing iteself" if and only if my Windows 10 is hibernating

kar200, 2020-07-24:
I actually reinstalled in my case but thanks for the info. I have foolowed @dubagunta-rohit advice and turned it off that way.

CDLLY, 2021-02-03:
I had the same problem after installing Mac OS, but it worked well with just Windows 10 20H1，the chromeos.img is stored in windows partition.