sebanc/brunch #419: Sleep when cover is closed not working

Shamw0w, 2020-07-22:
Hi, I have a Dell Inspiron 3000. In power settings, I have it set when idle to turn off display when charging and to sleep when on battery, and have activated the slider to sleep when cover is closed. However, when I close the lid it does not sleep. I have the computer running the built in screen and an external monitor (via HDMI out); when I shut the lid the external monitor stays on and ChromeOS remains unlocked - the content that was on  my built in screen jumps to the external monitor and I can still  interact with my mouse. When I open the lid, the built in screen remains off for a fraction, it turn turns on and the content then shifts back from the external monitor to the built in, and what was previously on the external monitor comes back. Any advice on how I can get sleep to work?

sebanc, 2020-07-23:
Hi, sorry but I don't think I can do something about that as it is ChromeOS current behavior:
https://support.google.com/chromebook/thread/23908010?hl=en

Shamw0w, 2020-07-23:
Oh, sorry didn't realise it was a known bug in ChromeOS. I guess if I
unplug the HDMI then it will sleep when I shut the lid?

closed 2020-07-27 by Shamw0w