sebanc/brunch #420: Chrome OS Windows 10 dual boot on Asus Zenbook ux36f2fa

EhsanulH, 2020-07-23:
Hello,
Everything I tested so far works correctly except for the camera. The webcam is listed as "usb2.0 hd ir uvc webcam" in the windows device manager. Is there any solution to this?

sebanc, 2020-07-23:
Hi, "reef" recovery image currently has the best camera support, could you try it ?

EhsanulH, 2020-07-23:
> Hi, "reef" recovery image currently has the best camera support, could you try it ?

Can you please provide the link?

Crescendo-BLYAT, 2020-07-23:
> > Hi, "reef" recovery image currently has the best camera support, could you try it ?
> 
> Can you please provide the link?

https://cros-updates-serving.appspot.com/

search for reef

EhsanulH, 2020-07-23:
> > > Hi, "reef" recovery image currently has the best camera support, could you try it ?
> > 
> > 
> > Can you please provide the link?
> 
> https://cros-updates-serving.appspot.com/
> 
> search for reef

Thanks, I am actually a noob. Can you please tell me how to change the recovery from rammus to reef? Will it erase my files?

rayqusa, 2020-08-05:
please clean install, it is very easy, dont be big baby

yeah, your data going to lose, simply put on cloud.