sebanc/brunch #421: Upated to chrome os 84 via OTA upate with brunch 83 working well with no hiccups

b1t0y, 2020-07-23:
This is not an issue.  Just informing that I updated to chrome os 84 via OTA update with brunch 83 still working well.

jhedfors, 2020-07-23:
Same here, I took the leap. :)
I am wondering if it is possible/makes sense for when @sebanc has built in compatibility for beta/dev builds, that the brunch release name can reflect that compatible build number?

dubagunta-rohit, 2020-07-23:
Thank you so much @b1t0y

massatt212, 2020-07-23:
how do you guys update, I'm using rammus and reef but when I put it on beta or dev there is no update only ver.83

Skawtnyc, 2020-07-23:
> how do you guys update, I'm using rammus and reef but when I put it on beta or dev there is no update only ver.83

You need to enable updates in grub options. Instructions are on the main page here.

massatt212, 2020-07-23:
ok

ragnoskun, 2020-07-24:
I need help, after update i dont have audio ?? After update, install brunch ( latest stable build )and again no sound,please help ( sorry for bad eng )

NSbuilder, 2020-07-24:
Does anyone knows when Brunch 84 stable comes out? ( Currently Chrome OS 84 firmwares are not available yet 24/07 )

dubagunta-rohit, 2020-07-24:
Chrome os84 is stable on the lastest brunch... Just Update the Chrome OS  over the air.

b1t0y, 2020-07-24:
> I need help, after update i dont have audio ?? After update, install brunch ( latest stable build )and again no sound,please help ( sorry for bad eng )

Try muting the speaker first then restart chrome os.  Unmute it and see if sound works.

NSbuilder, 2020-07-24:
> Chrome os84 is stable on the lastest brunch... Just Update the Chrome OS over the air.

idk, @ragnoskun just said he has audio problem..

ragnoskun, 2020-07-25:
@b1t0y still not working, no audio from stereo or headphones...

E1ZD, 2020-07-25:
@ragnoskun same here...
Audio NOT working, headphones or speaker...
Android apps NOT working, just spinning  and 
that's all... @b1t0y i tried that and nothing

NSbuilder, 2020-07-25:
We know that the update is not bug safe. Maybe change the misleading headline and provide specs for our computers.

Me: Lenovo Y510P

jhedfors, 2020-07-25:
> We know that the update is not bug safe. Maybe change the misleading headline and provide specs for our computers.
> 
> Me: Lenovo Y510P

@NSbuilder This post was @b1t0y 's experience. Nothing misleading about it at all. 

If you have an issue with your setup, maybe you should create your own post. :)

b1t0y, 2020-07-26:
> We know that the update is not bug safe. Maybe change the misleading headline and provide specs for our computers.
> 
> Me: Lenovo Y510P

I updated 6 laptops/pc and all worked well without hiccups: 

1. Lenovo u410 touch
2. Lenovo u430 touch
3. Dell inspiron 7347 touch
4. Intel nuc i3 
5. Hp envy recline 27 AIO
6. Asus x55m 15.6 

It did worked for me so I just shared my experience.  
Maybe you could try using reef or reinstalling it if would work. Better yet wait for @sebanc brunch 84 release. I just suggested the mute and unmute because it worked before in chromefy/croissant.

quakeaz, 2020-07-29:
Working without issue after OTA update on an Acer C710 Chromebook (Brunch 83 v4/CrOS 84)

closed 2020-08-05 by b1t0y