sebanc/brunch #422: Virtual Desktop Gestures?

jhedfors, 2020-07-24:
I have been trying to figure out how to use the Virtual Desktop feature, and while I have found some keyboard shortcuts, I would like to use touchpad guestures where you should be able to 4-finger swipe. I see documentation to enable this flag:

chrome://flags/#enable-virtual-desks-gestures

...but that does not exist.

Separately, is there a shortcut to the overview screen, as we do not have the Chromebook overview button?

dubagunta-rohit, 2020-07-24:
f5 on mine asus vivobook x507uar

jhedfors, 2020-07-24:
> f5 on mine asus vivobook x507uar

Thanks!. Fn+F5 worked for me - I had not tried the Fn+ keys.

efbe2121, 2020-07-25:
Here's what i found so far
F1 F2 : Backward Forward
F3 : Refresh
F4 : Full Screen
F5 : Overview Screen (what youre looking for)
F6 F7 : Brightness control
F8 F9 F10 : Volume control

jhedfors, 2020-07-25:
@efbe2121 Thanks. I can confirm all those work for me as well when selected with Fn key. I know I can change that in BIOS but then I will need to press Fn+ to use the media controls. For now, using the Fn+F5 works for accessing the Overview Screen.

Crescendo-BLYAT, 2020-07-26:
no need to change it in BIOS, just go to Chrome OS settings > device > keyboard > (there is a toggle) Treat top-row keys as function keys.

jhedfors, 2020-07-26:
> no need to change it in BIOS, just go to Chrome OS settings > device > keyboard > (there is a toggle) Treat top-row keys as function keys.
Thanks, however I don't think that is what that toggle actually does. This toggle just re-maps the modifier key from Fn+ to Search+. In my case (since I do not have a hardware search key) it simply disabled access to the alternate functions.

ayushbathrey, 2020-08-01:
search + f5 works for me

closed 2020-10-29 by jhedfors