sebanc/brunch #423: Minor Issues With the Surface 3 (non-pro)

ShadowcraftDesign, 2020-07-24:
I was surprised at how well everything works out of the box, but I have two problems. I am using the latest stable version (20200708) with rammus 83.
1. Wired audio is not working. When I plug headphones in they are detected and the audio output is switched, but I get no sound no matter what the volume is. Bluetooth audio and the speakers work fine.
2. Poor battery while sleeping. While in use the battery is excellent, but I was loosing 25% overnight. I added suspend_s3 in grub, but I haven't seen any difference.

I am also having trouble booting from grub, I added the menuentry but when I boot it it can't find the partition the image is on. I am pretty sure this is unrelated to brunch as it works just fine when I added the same menuentry to the USB.

choo0166, 2020-07-27:
The headphone channel is muted by default, to change this open crosh shell, do `sudo nano /usr/share/alsa/ucm/chtrt5645/HiFi.conf`.
Edit the lines `cset "name='Headphone Switch' off"` and `cset "name='Headphone Channel Switch' off"` to 'on'.
Warning: The headphone volume is extremely loud by default, so turn it all the way down to avoid blowing your eardrums :)

ShadowcraftDesign, 2020-07-27:
Wired audio works now, thanks!

daviddean99, 2020-09-01:
Once you have booted ChromeOS from USB, use this command to install directly onto the Surface 3 internal HDD:

ctr-alt-t to open terminal
shell
sudo chromeos-install -dst /dev/mmcblk0

When you restart and remove the USB drive then it should find the partitions just fine.