sebanc/brunch #424: alc5672 Realtek audio not supported

gtaws, 2020-07-26:
Using a Dell Venue Pro 5055, and it seems like there are not drivers for Realtek ALC5672, as the only audio output supported so far is HDMI out. Any hints on how to resolve this?

jhgxgAHS, 2020-07-26:
i also have a Dell laptop inspiron 5458 with realtek and microphone is not working and sometimes sound also feels very low.

gtaws, 2020-07-26:
I guess looking around I can't see if there are updated alsa drivers from realtek that can be compiled for brunch kernel. Can someone help?

jwalcek, 2020-07-27:
I have a Venue 8 Pro (same CPU I think) without sound as well. I've been meaning to add the Bay Trail option to the framework options, but haven't had time. Did you try passing options=baytrail_chromebook to the kernel command line?

gtaws, 2020-07-28:
@jwalcek I just did, and although it didn't solve my audio issue, touchscreen seems to work better. thanks for the tip anyways.

tornadox, 2020-07-28:
The options=baytrail_chromebook works for only one certain options
https://github.com/sebanc/brunch/issues/201

jafer241177, 2020-08-09:
also lenovo idealpad  flex 3 sound very low

closed 2026-04-20 by github-actions[bot]