sebanc/brunch #426: Android apps not working after update with Brunch r83 k4.19 stable 20200708

jhgxgAHS, 2020-07-26:
I had brunch 81 stable . after updating my android apps ar not working . its just showing loading . my laptop is a dell inspiron 5458 with a 5th gen core i3 prossesor

efbe2121, 2020-07-26:
Did you wait before you login? Like 1 or 2 minutes.

jhgxgAHS, 2020-07-28:
No , I didn't .

efbe2121, 2020-07-28:
> No , I didn't .
> […](#)

That's the reason why it happened :)
Based on what sebanc ever said iirc, you need to wait about 1 or 2 minutes at login screen instead login in immediately because the Android component is loading after login UI appeared.

quakeaz, 2020-07-29:
> > No , I didn't .
> > […](#)
> 
> That's the reason why it happened :)
> Based on what sebanc ever said iirc, you need to wait about 1 or 2 minutes at login screen instead login in immediately because the Android component is loading after login UI appeared.

I have this issue as well. Most of the time waiting a couple minutes before logging in seems to work, but other times it seems the container doesn't start properly and the apps will try to load forever (loading circle over the icon in dock). Maybe my computer is just slow and I need to wait longer then 1-2 mins before logging in, I'll try and see.

efbe2121, 2020-07-29:
It works perfectly fine with me, im at ver 85 already and android apps are working perfectly fine when i waited about 1 or 2 minutes on login screen.

Skawtnyc, 2020-07-30:
Android actually doesn't finish loading until AFTER you log in. There are some internal init scripts in Android that only execute after the user has logged in. Log in right away, then wait a minute or two before launching Android apps. After you get them running the first time, the next time you reboot it should be faster.

efbe2121, 2020-07-30:
I've tried that aswell, i wait for about more than 5 minutes after i login right away. It just not working at all

closed 2026-04-20 by github-actions[bot]