sebanc/brunch #430: Chrome OS suddenly crashes and restarts

skandavc18, 2020-07-27:
I am on chrome OS 84 in Lenovo yoga 730. I have enabled updates, hyperthreading, and auto-brightness.  There are sudden and unexpected crashes. There are crashes even in Chrome OS 83 without hyperthreading and auto-brightness. I am using the latest brunch framework.

Please help.

the-quantum-nargle, 2020-07-31:
I have the same issue, the reboots reduced a lot when I downgraded to **Brunch r83 k4.19 testing 20200704**
But they're still there and random. I have an HP Pavilion x360, with i7-8550U, NVME 500gb, 12 GB Ram, Nvidia MX130 (but blacklisted from grub). 

ChromeOS is installed along side of Arch Linux and Windows 10 currently. on a 68GB partition. (tried both ext4 and ntfs)

I went through /var/log/messages and found some lines which might indicate why, but I'm not entirely sure what they mean

1. 
`2020-07-30T15:22:24.837056+05:30 ERR VM(3)[10508]: message repeated 8 times: [  [devices/src/virtio/net.rs:240] net: tx: failed to write frame to tap: File descriptor >
2020-07-30T15:22:27.550519+05:30 INFO VM(3)[10502]: [  632.881453] maitred: Detaching loopback devices
2020-07-30T15:22:27.550736+05:30 INFO VM(3)[10502]: [  632.881839] maitred: Removing device mapper devices
2020-07-30T15:22:27.550814+05:30 INFO VM(3)[10502]: [  632.881975] maitred: Unmounting filesystems
2020-07-30T15:22:27.551035+05:30 INFO VM(3)[10502]: [  632.882184] maitred: Unmounting /mnt/stateful/lxd/devlxd
2020-07-30T15:22:27.551146+05:30 INFO VM(3)[10502]: [  632.882265] maitred: Unmounting /mnt/stateful/lxd/shmounts
2020-07-30T15:22:27.551227+05:30 INFO VM(3)[10502]: [  632.882371] maitred: Unmounting /mnt/stateful
2020-07-30T15:22:27.551371+05:30 INFO VM(3)[10502]: [  632.882507] maitred: Failed to unmount /mnt/stateful: Device or resource busy (16)
2020-07-30T15:22:27.551426+05:30 INFO VM(3)[10502]: [  632.882573] maitred: Unmounting /mnt/shared
2020-07-30T15:22:27.551507+05:30 INFO VM(3)[10502]: [  632.882659] maitred: Unmounting /opt/google/cros-containers
2020-07-30T15:22:27.551641+05:30 INFO VM(3)[10502]: [  632.882800] maitred: Unmounting /var
2020-07-30T15:22:27.574394+05:30 ERR 9s[10491]: error while handling client 3:2305086445: failed to fill whole buffer
2020-07-30T15:22:27.588823+05:30 INFO VM(3)[10502]: [  632.919733] Unregister pv shared memory for cpu 2
2020-07-30T15:22:27.588894+05:30 INFO VM(3)[10502]: [  632.919734] Unregister pv shared memory for cpu 0
2020-07-30T15:22:27.588930+05:30 INFO VM(3)[10502]: [  632.919741] Unregister pv shared memory for cpu 5
2020-07-30T15:22:27.588986+05:30 INFO VM(3)[10502]: [  632.919742] Unregister pv shared memory for cpu 3
2020-07-30T15:22:27.589024+05:30 INFO VM(3)[10502]: [  632.919744] Unregister pv shared memory for cpu 4
2020-07-30T15:22:27.589067+05:30 INFO VM(3)[10502]: [  632.919745] Unregister pv shared memory for cpu 6
2020-07-30T15:22:27.589121+05:30 INFO VM(3)[10502]: [  632.919747] Unregister pv shared memory for cpu 7
2020-07-30T15:22:27.589152+05:30 INFO VM(3)[10502]: [  632.919989] Unregister pv shared memory for cpu 1
2020-07-30T15:22:27.589180+05:30 INFO VM(3)[10502]: [  632.920075] kvm: exiting hardware virtualization
2020-07-30T15:22:27.590115+05:30 INFO VM(3)[10502]: [  632.921091] reboot: Restarting system
2020-07-30T15:22:27.590191+05:30 INFO VM(3)[10502]: [  632.921134] reboot: machine restart
2020-07-30T15:22:27.591540+05:30 INFO VM(3)[10498]:  [src/linux.rs:1854] vcpu requested shutdown
2020-07-30T15:22:27.751579+05:30 INFO VM(3)[10498]:  [src/main.rs:1423] crosvm has exited normally
2020-07-30T15:22:27.965397+05:30 INFO vm_cicerone[10472]: Received NotifyVmStopped request
2020-07-30T15:22:27.965486+05:30 WARNING vm_cicerone[10472]: Latent container left in VM termina of penguin
2020-07-30T15:22:27.966914+05:30 INFO seneschal[10447]: Received request to stop server
2020-07-30T15:22:27.973343+05:30 INFO seneschal[10447]: Process 2 killed by signal 15
2020-07-30T15:22:27.973795+05:30 ERR seneschal[10447]: Unable to unmount path; not deleting runtime directory: Invalid argument (22)`

2. (messages files is mostly filled with this, but I don't think its a problem, just a warning? because my rtl Wifi+Bluetooth works just fine)
`2020-07-30T15:20:32.745155+05:30 ERR kernel: [  683.828866] BUG: sleeping function called from invalid context at drivers/net/wireless/realtek/rtw88/fw.c:235
2020-07-30T15:20:32.745201+05:30 ERR kernel: [  683.828893] in_atomic(): 1, irqs_disabled(): 0, pid: 154, name: kworker/u16:5
2020-07-30T15:20:32.745219+05:30 WARNING kernel: [  683.828914] CPU: 1 PID: 154 Comm: kworker/u16:5 Tainted: G WC 4.19.122-brunch-sebanc #1unch-sebanc #1
2020-07-30T15:20:32.745228+05:30 WARNING kernel: [  683.828930] Hardware name: HP HP Pavilion x360 Convertible 14-cd0xxx/8487, BIOS F.35 11/18/2019
2020-07-30T15:20:32.745235+05:30 WARNING kernel: [  683.828961] Workqueue: phy0 rtw_watch_dog_work [rtw88_core]
2020-07-30T15:20:32.745240+05:30 WARNING kernel: [  683.828973] Call Trace:
2020-07-30T15:20:32.745245+05:30 WARNING kernel: [  683.828997]  dump_stack+0x64/0x7c
2020-07-30T15:20:32.745250+05:30 WARNING kernel: [  683.829015]  ___might_sleep+0x114/0x129
2020-07-30T15:20:32.745256+05:30 WARNING kernel: [  683.829038]  rtw_fw_send_h2c_command+0xa9/0x18d [rtw88_core]
2020-07-30T15:20:32.745261+05:30 WARNING kernel: [  683.829063]  rtw_fw_set_pwr_mode+0x8f/0xa9 [rtw88_core]
2020-07-30T15:20:32.745267+05:30 WARNING kernel: [  683.829086]  rtw_leave_lps+0x8e/0x118 [rtw88_core]
2020-07-30T15:20:32.745272+05:30 WARNING kernel: [  683.829109]  rtw_watch_dog_work+0x194/0x209 [rtw88_core]
2020-07-30T15:20:32.745277+05:30 WARNING kernel: [  683.829127]  process_one_work+0x1ac/0x2c1
2020-07-30T15:20:32.745281+05:30 WARNING kernel: [  683.829142]  worker_thread+0x1a3/0x252
2020-07-30T15:20:32.745310+05:30 WARNING kernel: [  683.829156]  ? rescuer_thread+0x2c5/0x2c5
2020-07-30T15:20:32.745315+05:30 WARNING kernel: [  683.829173]  kthread+0xf3/0xf8
2020-07-30T15:20:32.745320+05:30 WARNING kernel: [  683.829190]  ? kthread_delayed_work_timer_fn+0x77/0x77
2020-07-30T15:20:32.745324+05:30 WARNING kernel: [  683.829216]  ret_from_fork+0x35/0x40`

skandavc18, 2020-07-31:
@the-quantum-nargle  I downgraded to Brunch r81 k4.19 stable 20200510 and it's working completely fine even though I am still on Chrome OS 84. No sudden crashes and reboots now.

the-quantum-nargle, 2020-07-31:
I can't go to brunch 81 since I use a kernel parameter for alternative touchpad which was introduced in brunch 83, however, there are 2 more brunch83 testing releases which I can downgrade further to, so I'll give them a try and post here. Thanks for the suggestion.

sebanc, 2020-09-05:
the new r85 build should solve that issue

skandavc18, 2020-09-16:
yes. working properly. Thanks

closed 2020-09-16 by skandavc18