sebanc/brunch #431: Stylus not working Chuwi HI10 X

Ebbot9876, 2020-07-27:
My stylus equipped with Chuwi HI10 is not working in Chrome OS.
It´s detected with evtest but not working in UI.
[evtest.txt](https://github.com/sebanc/brunch/files/4982290/evtest.txt)
[lsmod.txt](https://github.com/sebanc/brunch/files/4982291/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4982292/dmesg.txt)

sebanc, 2020-07-28:
Which recovery image do you use ?

Ebbot9876, 2020-07-28:
Rammus now 84, tried octopus and reef!

sebanc, 2020-07-28:
Could you try to add "--has-internal-stylus" to /etc/chrome_dev.conf ?

Ebbot9876, 2020-07-28:
I have already tried!

sebanc, 2020-07-28:
Could you look for the "Pen" input device path in your dmesg and post the output of `udevadm info <path_to_input_device_sysfs>` ?

Ebbot9876, 2020-07-28:
I don`t know if this is right?...
udevadm info -q path /dev/input/event26
/devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-9/i2c-GXTP7386:00/0018:27C6:011A.0005/input/input26/event26

sebanc, 2020-07-29:
Try `udevadm info /sys/devices/pci0000:00/0000:00:17.0/i2c_designware.4/i2c-9/i2c-GXTP7386:00/0018:27C6:011A.0005/input/input26/event26`

Ebbot9876, 2020-07-29:
Hi, here is output from udevadm!
Path has changed.
[udevadm.txt](https://github.com/sebanc/brunch/files/4993328/udevadm.txt)

Ebbot9876, 2020-08-20:
@sebanc I tried latest FydeOS with 5.4 kernel, stylus works fine there!
Is it possible to use 5.4 kernel with brunch?

sebanc, 2020-08-29:
I have added you to the brunch-next repo which has 5.4 kernel so that you can test

SeanJean09, 2020-12-10:
any updates with this? I updated my Chuwi Hi10X to the latest brunch release and updated chrome to 87, here are some of my issues:

- stylus not working
- screen rotated to portrait, even on startup (has to set my Display to 90 degrees on Settings)
- rotation sensor not working
- camera rotated to portrait
- front camera not working

sebanc, 2020-12-11:
I think the screen and camera rotation had been fixed in a previous versions but I forgot to include the patch in the 5.4 kernel, could you confirm that they work correctly with the previous release (Brunch r86 k4.19 stable 20201014) ? I will add the patch in 5.4 kernel.

Only Intel hid sensors have been ported to work with brunch and your device use different ones so rotation will not work.

To be honest I have no idea why stylus won't work on this device, it is the only issue of this type that I have seen.

Chromeos OS r87 seems to have camera issues.

Globally, for now I would suggest to use Brunch r86 k4.19 stable 20201014 and chromeos r86.

Ebbot9876, 2020-12-11:
I tested r86 4.19 version about two months ago. Then screen turned upside down.

SeanJean09, 2020-12-11:
> I think the screen and camera rotation had been fixed in a previous versions but I forgot to include the patch in the 5.4 kernel, could you confirm that they work correctly with the previous release [...]

no, the screen is upside down. have to set my screen to 180 degrees that time. same with the camera. I'll still use this update though as I'm not using the camera that much. I just wosh there's something we can do to fix the stylus issue, as I mainly use this as a whiteboard for my lectures

SeanJean09, 2020-12-13:
> I uploaded a build which should hopefully fix screen and camera orientation in the "brunch-testing" repo.
> 
> For the stylus, I looked at your old evtest log and it seems that chromeos does not enable input devices if they do not provide ABS_X and ABS_Y values which is the case here:
> 
> ```
> bool EventDeviceInfo::HasTablet() const {
>   return HasAbsXY() && HasPointer() && HasStylus();
> }
> ```
> 
> in https://source.chromium.org/chromium/chromium/src/+/master:ui/events/ozone/evdev/event_device_info.cc
> 
> There is not much I can do about this, if there is a bios/touchscreen firmware for your HI 10 you can try to apply it.

I've tested the unstable build and the screen is fixed. thanks for that! the camera is in the right orientation as well, but it's flipped/mirrored, but it's fine since we can flip it again after taking a picture.

The firmware of this specific tablet is only Windows-based, and the manufacturers never made a way for us users to use other Linux-based OS. I tried Linux Mint and PopOS before and they seem to have some form of support on the Pen/Stylus Input for this. I just don't know how I can use their implementation for Chrome OS

sebanc, 2020-12-13:
Mint and PopOS just use libinput which does not care about ABS_X and ABS_Y, unfortunately there is no way to replace the chromeos touch/stylus driver.

robs421, 2021-01-01:
Hi, i am a noob with regards brunch and Chromeos installation. I am running a mercer guru laptop with celeron chipset, i have win 10 pro on the internal 64gb drive and chromeos running on a 256gb m.2 drive. I have a similar problem as Ebbot, everything works fine. Only there is no stylus support. I read through the above, but i dont know how to apply the fix suggested by editing the chrome_dev file. Would you mind explaining in layman terms? Thank you.

ccchan234, 2023-03-30:
any news in 2023? thx