sebanc/brunch #433: unmount: can't unmount /dev: Device or ressource busy

dmace7, 2020-07-27:
Hello, when I boot on `Chrome Os` there is a dark window with white text. Everything goes well until an error appears : ```unmount: can't unmount /dev: Device or ressource busy```. Then the computer restart reboot on `Chrome Os` and the same thing happens over and over. Any hint. Thanks in advance...

sebanc, 2020-07-28:
Hi, I don't really see what you mean by "there is a dark window with white text", could you attach a photo of your screen when it happens ?

dmace7, 2020-07-28:
[image] It's something like that.

ayushbathrey, 2020-08-02:
> [image] It's something like that.
It usually happens if Windows 10 fast startup is enabled.

dmace7, 2020-08-02:
Hello, thank you for your answer, how do I disable  `Windows fast startup mode` ? Thanks in advance

sebanc, 2020-08-29:
If you are still interested, I will need a short video of the boot process to understand what the issue is.

sebanc, 2020-08-29:
And also your laptop model / specs

closed 2026-04-16 by github-actions[bot]