sebanc/brunch #434: Videos not working

Sahib911, 2020-07-28:
Videos on any site(youtube) are not working .

Crescendo-BLYAT, 2020-07-30:
detailed info please?
laptop model, image used, brunch version, what did you do to recreate the issue...

Sahib911, 2020-07-31:
Laptop Hp Pavilion g6 2230tx
latest brunch (it didnt work in older version either)
Image is Sammus 
When the Os is updated & booted for the first time it works flawlessly but after restarting videos doesn't work .
In #252 Sebanc said the issue is related to a sound patch that he will revert in the next build .

Sahib911, 2020-09-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/5177572/dmesg.txt)

sebanc, 2020-09-05:
Does adding "snd_hda_intel.index=0" to the kernel command line helps ?

Sahib911, 2020-09-05:
[dmesg.txt](https://github.com/sebanc/brunch/files/5178700/dmesg.txt)
No Luck

sebanc, 2020-09-06:
Could you try the below solution:
https://forums.linuxmint.com/viewtopic.php?p=589566&sid=f88214712806e689c134fbd3baa3ee00#p589566

Sahib911, 2020-09-06:
Didn't work . 
[dmesg.txt](https://github.com/sebanc/brunch/files/5179215/dmesg.txt)

closed 2026-04-16 by github-actions[bot]