sebanc/brunch #435: Screen rotation - Lenovo 300e 2nd gen Windows

markinhosmrk, 2020-07-28:
I am using brunch on this laptop (Lenovo 300e 2nd gen Intel Windows version), using the recovery image for octopus (for 300e Chromebook since I guess HW is mostly the same and it could support the stylus).

Screen rotation is not working. I see the ish-hid that should be supported, as below:
`[769.065770] ish-hid {33AECD58-B679-4E54-9BD9-A04D34F0C226}: [hid-ish]: enum_devices_done OK, num_hid_devices=2
`

Any tips on what to check?

Thanks!
Marcos

sebanc, 2020-07-29:
You have to be in tablet mode for the device to rotate and tablet mode detection is not available on many devices. Try pressing CTRL + ALT + SHIFT + T to trigger it manually.

Skawtnyc, 2020-07-29:
> You have to be in tablet mode for the device to rotate and tablet mode detection is not available on many devices. Try pressing CTRL + ALT + SHIFT + T to trigger it manually.

You need to enable the debug shortcuts flag to be able to use that switch:

chrome://flags/#ash-debug-shortcuts

markinhosmrk, 2020-08-02:
Indeed, with tablet mode screen rotation works. I'll investigate on the tablet mode detection! Thanks!

rayqusa, 2020-08-05:
where to write debug flag? at chrono? or EFI grub?