sebanc/brunch #436: Suspend stops touchscreen - Lenovo 300e 2nd gen Windows

markinhosmrk, 2020-07-29:
Hi, another issue!
After resume from suspend the touchscreen stops working. This also happens with Linux and I have a fix working on ChromeOS: unbind then bind again the device.

```
echo -n "i2c-ELAN238E:00" | tee /sys/bus/i2c/drivers/i2c_hid/unbind
echo -n "i2c-ELAN238E:00" | tee /sys/bus/i2c/drivers/i2c_hid/bind
```

On Linux I use a systemd system-sleep script to do it; not sure how to do it on ChromeOS. I was checking D-Bus events generated and there are a couple when entering suspend state but did not found one for leaving:

```
signal time=1595980361.558590 sender=:1.150 -> destination=(null destination) serial=662 path=/org/chromium/PowerManager; interface=org.chromium.PowerM
anager; member=SuspendImminent
   array of bytes [
      08 82 80 c8 f6 02 10 01
   ]
   
signal time=1595980764.384896 sender=:1.150 -> destination=(null destination) serial=1095 path=/org/chromium/PowerManager; interface=org.chromium.PowerManager; member=SuspendDone

```
Thanks!
Marcos

sebanc, 2020-07-29:
Hi,

Unfortunately I was also not able to find a resume hook in chromeos. I would be interested to know if you find one.

markinhosmrk, 2020-08-02:
I managed to make it work with dbus-monitor:

`dbus-monitor --system --profile "interface='org.chromium.PowerManager',member='SuspendDone'"`

When this signal comes it means it has woken up from suspend, then I run the commands I need. I set it up as a daemon.

closed 2020-08-02 by markinhosmrk

bybenjamin, 2022-02-05:
> dbus-monitor ile çalışmasını sağladım:
> 
> `dbus-monitor --system --profile "interface='org.chromium.PowerManager',member='SuspendDone'"`
> 
> Bu sinyal geldiğinde beklemeden uyanmış demektir, sonra ihtiyacım olan komutları çalıştırıyorum. Bir daemon olarak kurdum.

Hello markin
First of all, thank you for this project.
Everything works great..
The only problem for me is the touchpad stops working after sleep.
I am using brunch 97 and coral 97.
My touchpad is ''i2c_hid i2c-ELAN1200''
I made a script to make it work again.
my script '' rmmod i2c_hid ; modprobe i2c_hid ''
Is there a way to automate this after sleep?
Or how can I prevent the i2c_hid module from going to sleep?
Thank you for your help... :)

sebanc, 2022-02-05:
I think I more or less understood this.

If you create a file /etc/init/touchpad-resume.conf with the below content, does the touchpad resume correctly ?

```
start on stopped udev-trigger

script
dbus-monitor --system --profile "interface='org.chromium.PowerManager',member='SuspendDone'" | while read -r line; do
	rmmod i2c_hid
	modprobe i2c_hid
done
end script
```

bybenjamin, 2022-02-06:
> I think I more or less understood this.
>
> If you create a file /etc/init/touchpad-resume.conf with the below content, does the touchpad resume correctly ?
>
> ```
> start on stopped udev-trigger
>
> script [...]

Thanks sebanc.. it is  perfect work :)