sebanc/brunch #438: Intel 10th Gen - Intel Iris Plus Graphics Chrome OS not booting and stuck

ganesh8592, 2020-08-01:
my laptop details are: hp spectre x360 13t late 2019 with i7 1065G7 and intel iris Graphics 

can you please provide kernel 5.4 with specific repo

ganesh8592, 2020-08-02:
got successful installation

closed 2020-08-02 by ganesh8592

Namaneo, 2020-08-16:
Exact same issue here, and exact same configuration on an recent XPS 13. How did you manage to fix this? Manually replacing the kernel inside the image, or something else?

ganesh8592, 2020-08-16:
tomorrow I will provide link with related all files

Namaneo, 2020-08-16:
That would be fantastic ! Thank you 🙂

ganesh8592, 2020-08-18:
hi bro,

sry  for delay

intel 10th gen need kernel 5.4 or ablove

brunch: Brunch r83 k5.4 unstable 20200711
<https://github.com/sebanc/brunch-next/releases/tag/r83-k5.4-unstable-20200711>
link:
https://github.com/sebanc/brunch-next/releases/tag/r83-k5.4-unstable-20200711
developer/supported by https://github.com/sebanc

thanks & regards
Ganesh

Namaneo, 2020-08-18:
Thank you 🙂 It seems this repository is private.

@sebanc Is it possible to have the permissions to access it? 
And when do you think the update we will moved to the stable repo?

Thanks !

sebanc, 2020-08-18:
I happened to catch this one by chance but you are commenting on a closed issue. Please use issue #18 instead.

@Namaneo you have received an invite to brunch next, however it is not planned to be merged to stable for now as kernel 5.4  is currently not compatible with many devices with Intel gen < Skylake.

ryansg8, 2020-11-23:
@sebanc Sorry for the spam you must get, but I am in the process of trying to install this onto my Ice Lake laptop, and access to brunch-next branch for kernel 5.4 would be awesome so I can get it booting. Thanks!