sebanc/brunch #440: Solution to enable touchpad gestures on Synaptic devices/Request To Force Clamshell Mode

CarlitoisLoco, 2020-08-02:
The touchpad gestures were not working on my Skylake HP Pavilion x360 with a Synaptics touchpad and I found that it can by easily solved by enabling 'psmouse.synaptics_intertouch=1' in the grub config 

(

> options= [...]<space>psmouse.synaptics_intertouch=1

)

Found the solution from [this guide](https://docs.google.com/document/u/1/d/e/2PACX-1vSV4dCvvx6aZmOeMWDmrT_1GwtEJlkzRk12E8nkNL5VaesMr55Hk09neKbaHu-gsXQlM6SwKLiRnHPK/pub)

But I found that this consequently forces Tablet Mode to be enabled at all times without there being a way to revert back to clamshell Mode since ctrl+shift+alt+t doesn't work after enabling it (and the chrome flag to force disable tablet mode has been removed for some time now)

It'd be really great if there's a way to force enable clamshell mode since tablet mode is really inconvenient but I like the gestures enough to be forced to use tablet mode for the time being

ayushbathrey, 2020-08-02:
I also have same touchpad, but "psmouse.synaptics_intertouch=1" doesn't work for me, also touchpad doesn't work that smoothly and 3-4 finger gesture won't work.

CarlitoisLoco, 2020-08-02:
It should normally work if you have a Synaptics touchpad. Also have you tried alt_touchpad_config or alt_touchpad_config2?

ayushbathrey, 2020-08-02:
Yes I tried but without psmouse...* option.

ayushbathrey, 2020-08-02:
Yup I tried all of them, but no luck, also I was not forced to use tablet mode.

CorvetteCole, 2020-12-29:
I have this same issue :(

CorvetteCole, 2020-12-29:
@CarlitoisLoco did you ever find a workaround for this?

sebanc, 2020-12-30:
I have no idea how this parameter can trigger tablet mode. Do you have "hp_wmi"module loaded (type "lsmod" in crosh shell to check). If so, could you try booting with "module_blacklist=hp_wmi" added to the kernel command line ?

sebanc, 2020-12-30:
I might have found something:
https://patchwork.kernel.org/project/platform-driver-x86/patch/20201207165914.414091-1-hdegoede@redhat.com/

sebanc, 2020-12-30:
Could you also try "module_blacklist=intel_vbtn" ?

CarlitoisLoco, 2021-01-02:
> @CarlitoisLoco did you ever find a workaround for this?

Hey! It has been working fine for me since updating to chromeos r86, which build are you running?

maxdojin, 2021-01-04:
I have the same problem
hp_wmi module is not loaded
I also tried module_blacklist = intel_vbtn

librehat, 2021-02-27:
I've been having the same issues since Chrome OS 87 update. I don't have `intel_vbtn` module loaded at all.

```
chronos@localhost / $ lsmod              
Module                  Size  Used by
vhost_vsock            20480  9
vmw_vsock_virtio_transport_common    24576  1 vhost_vsock
vsock                  36864  31 vmw_vsock_virtio_transport_common,vhost_vsock
vhost                  32768  1 vhost_vsock
rmi_smbus              16384  0
rmi_core               73728  1 rmi_smbus
intel_wmi_thunderbolt    16384  0
iio_trig_sysfs         16384  2
psmouse               135168  0
intel_xhci_usb_role_switch    16384  0
wmi                    24576  1 intel_wmi_thunderbolt
tpm_tis                16384  0
tpm_tis_core           20480  1 tpm_tis
acpi_pad               16384  0
btusb                  61440  0
btrtl                  20480  1 btusb
btbcm                  16384  1 btusb
btintel                24576  1 btusb
hid_sensor_cros_compat    16384  2
industrialio_triggered_buffer    16384  1 hid_sensor_cros_compat
kfifo_buf              16384  2 industrialio_triggered_buffer,hid_sensor_cros_compat
iwlmvm                307200  0
hid_sensor_iio_common    16384  1 hid_sensor_cros_compat
industrialio           65536  6 industrialio_triggered_buffer,kfifo_buf,iio_trig_sysfs,hid_sensor_cros_compat
snd_soc_skl           114688  0
snd_hda_codec_hdmi     57344  1
snd_soc_sst_dsp        16384  1 snd_soc_skl
mac80211              704512  1 iwlmvm
snd_hda_codec_conexant    20480  1
hid_sensor_hub         16384  2 hid_sensor_iio_common,hid_sensor_cros_compat
snd_soc_sst_ipc        16384  1 snd_soc_skl
snd_hda_ext_core       16384  1 snd_soc_skl
snd_hda_codec_generic    69632  1 snd_hda_codec_conexant
ledtrig_audio          16384  2 snd_hda_codec_generic,snd_hda_codec_conexant
snd_hda_intel          36864  3
iwlwifi               282624  1 iwlmvm
mei_hdcp               16384  0
mei_pxp                16384  0
snd_intel_dspcfg       24576  2 snd_hda_intel,snd_soc_skl
snd_hda_codec         110592  4 snd_hda_codec_generic,snd_hda_codec_conexant,snd_hda_codec_hdmi,snd_hda_intel
intel_lpss_pci         20480  0
mei_me                 24576  2
intel_lpss             16384  1 intel_lpss_pci
snd_hda_core           69632  7 snd_hda_codec_generic,snd_hda_codec_conexant,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_soc_skl
mei                    73728  5 mei_hdcp,mei_pxp,mei_me
idma64                 16384  0
snd_hwdep              16384  1 snd_hda_codec
intel_pch_thermal      16384  0
cfg80211              630784  3 iwlmvm,iwlwifi,mac80211
tablet_mode_switch     16384  0
i915                 1830912  50
video                  40960  1 i915
intel_gtt              20480  1 i915
i2c_algo_bit           16384  1 i915
drm_kms_helper        159744  1 i915
cec                    61440  2 drm_kms_helper,i915
drm                   397312  23 drm_kms_helper,i915
```

Any ideas? (Device: Lenovo YOGA 700)

closed 2021-03-16 by nadlax