sebanc/brunch #441: Request for bootloader hidden style or animation

hardshell92DE, 2020-08-02:
Hello everyone,

Chrome OS works great so far with on my Acer V3-372.

Currently i am playing with the grub.cfg in order to change the boot style to quiet.

**I would like to see no text on boot or a boot animation.**

I know that you can swap this with grub-customizer but i have not managed it on Chrome OS.

Unfortunately i dont have an idea how to change this. Please help.

CornishRattler, 2020-08-02:
Put "console=" after "cros debug 
That’ll give you a black screen with no text while booting

swebervna, 2020-09-24:
> Hello everyone,
> 
> Chrome OS works great so far on my Acer V3-372.
> 
> Currently i am playing with the grub.cfg in order to change the boot style to quiet.
> 
> **I would like to see no text on boot or a boot animation.**
> 
> I know that you can swap this with grub-customizer but i have not managed it on Chrome OS.
> 
> Unfortunately i dont have an idea how to change this. Please help.

I would love to do this - how do you go about editing the grub.cfg?

supechicken666, 2020-09-24:
> I would love to do this - how do you go about editing the grub.cfg?

To edit the grub.cfg, run this command in chrome os crosh shell by hitting ctrl+alt+t
sudo edit-grub-config

swebervna, 2020-09-24:
> > > Hello everyone,
> > > Chrome OS works great so far on my Acer V3-372.
> > > Currently i am playing with the grub.cfg in order to change the boot style to quiet.
> > > **I would like to see no text on boot or a boot animation.**
> > > I know that you can swap this with grub-customizer but i have not managed it on Chrome OS.
> > > Unfortunately i dont have an idea how to change this. Please help.
> > 
> > 
> > I would love to do this - how do you go about editing the grub.cfg?
> 
> To edit the grub.cfg, run this command in chrome os crosh shell by hitting ctrl+alt+t
> 
> ```shell
> sudo edit-grub-config
> ```

I'm in the Crosh shell and I got this result:

**crosh>** sudo edit-grub-config
ERROR: unknown command:  sudo

For the Crosh shell, isn't there something you have to do first to actually run these commands?

supechicken666, 2020-09-24:
Were you posted the question in reddit?
Just ready to close the tab and saw your question lol

swebervna, 2020-09-24:
> Were you posted the question in reddit?
> Just ready to close the tab and saw your question lol

Yes, that was me, haha!

swebervna, 2020-09-24:
Nevermind, I found. I gotta use "shell" first, it's all coming back to me

supechicken666, 2020-09-24:
> I'm in the Crosh shell and I got this result:
>
> **crosh>** sudo edit-grub-config
> ERROR: unknown command: sudo
>
> For the Crosh shell, isn't there something you have to do first to actually run these [...]

Type 
```shell
shell
``` 
before running the command, run this on crosh to enter shell first

swebervna, 2020-09-24:
Okay, I put "console=" after "cros_debug". For the "options=", could you clarify a bit more where I do that?

swebervna, 2020-09-24:
Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg

supechicken666, 2020-09-24:
> Okay, I put "console=" after "cros_debug". For the "options=", could you clarify a bit more where I do that?

By defluit, your grub.cfg won't have options line
Just copy my grub.cfg
```
set timeout=2

menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug enforce_hyperthreading=1 options=enable_updates
  initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}
```

supechicken666, 2020-09-24:
> Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg

Have you add a backspace between two options?

swebervna, 2020-09-24:
> > Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg
> 
> Have you add a backspace between two options?

No, they should have a space between each other.

supechicken666, 2020-09-24:
> > > Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg
> > 
> > 
> > Have you add a backspace between two options?
> 
> No, they should have a space between each other.
> 
> Maybe not though. Any chance I can edit the grub.cfg from the GRUB command line? Now I can't boot.

What method you install the chrome os?

swebervna, 2020-09-24:
> > > > Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg
> > > 
> > > 
> > > Have you add a backspace between two options?
> > 
> > 
> > No, they should have a space between each other.
> > Maybe not though. Any chance I can edit the grub.cfg from the GRUB command line? Now I can't boot.
> 
> What method you use to install the chrome os?

I followed this guide: https://beebom.com/how-install-chrome-os-on-pc/

I can just re install it all over again and follow directions from the Brunch's site, if I need to.

supechicken666, 2020-09-24:
> I followed this guide: https://beebom.com/how-install-chrome-os-on-pc/
>
> I can just re install it all over again and follow directions from the Brunch's site, if I need to.

Great! You can use the usb media you used to install chrome os to edit the file

supechicken666, 2020-09-24:
Once you get into the linux live desktop
Press `ctrl+alt+t` to open a terminal window
Then type the following command
```shell
mkdir ~/tmpm
sudo mount /dev/sda12 ~/tmpm
cd ~/tmpm
cd efi
cd boot
sudo apt install nano
sudo nano grub.cfg
```

supechicken666, 2020-09-24:
> I followed this guide: https://beebom.com/how-install-chrome-os-on-pc/
>
> I can just re install it all over again and follow directions from the Brunch's site, if I need to.

After doing above steps, you should able to edit the file again

swebervna, 2020-09-25:
> > > > > > Oh shoot, I somehow screwed up. Now my laptop is in a boot loop 😂 all I did was put "console=" after "cros_debug" in the grub.cfg
> > > > > 
> > > > > 
> > > > > Have you add a backspace between two options?
> > > > 
> > > > 
> > > > No, they should have a space between each other.
> > > > Maybe not though. Any chance I can edit the grub.cfg from the GRUB command line? Now I can't boot.
> > > 
> > > 
> > > What method you use to install the chrome os?
> > 
> > 
> > I followed this guide: https://beebom.com/how-install-chrome-os-on-pc/
> > I can just re install it all over again and follow directions from the Brunch's site, if I need to.
> 
> After doing above steps, you should be able to edit the file again

I tried reinstalling Chrome OS but now I'm getting this error when running the chromeos-install.sh script:
"/dev/nvme0n1" is not a valid disk name"
/dev/nvme0n1 is in fact the laptop's disk and it worked before, not sure why it won't now...

closed 2020-09-25 by hardshell92DE

supechicken666, 2020-09-25:
> I tried reinstalling Chrome OS but now I'm getting this error when running the chromeos-install.sh script:
> "/dev/nvme0n1" is not a valid disk name"
> /dev/nvme0n1 is in fact the laptop's disk and it [...]

This issue is closed, open a new issue and I will continue to answering you