sebanc/brunch #442: Mobile Network Option Doesn't Activate

sohawiz, 2020-08-02:
I'm Using Surface Go (1st Gen, 4G Model) and Successfully Installed Chrome OS on my tablet with Brunch Project.
Everything work well except mobile network. I'm using 4G Network with SIM card on Window 10, But on Chrome OS, I couldn't found any option to configure it. I only saw 'Wifi Network' Option.

Did Anybody Activate 4G Network in similar case? o couldn't I use Cellular on Chrome OS? 
Thanks.

CornishRattler, 2020-08-02:
Dell Latitude 5290 2in1 mobile network works fine on Chrome

sohawiz, 2020-08-03:
> Dell Latitude 5290 2in1 mobile network works fine on Chrome

Did you install current version(r83?) and didn't you configure anything to set up mobile network? 

Thank you for comment this issue

CornishRattler, 2020-08-03:
Current r83 and didn't need to configure anything

sohawiz, 2020-08-03:
> Current r83 and didn't need to configure anything
> [image]

Thanks Again, I can just find 'Wi-fi' option, I'll re-install chrome os

Rommelius, 2020-08-06:
I got the surface go as well. 

@CornishRattler 
can you let us know what LTE  WWAN card you have?

maybe its a driver issue for us @sohawiz

CornishRattler, 2020-08-06:
As I said I have a Dell 5290 not a Surface Go

sohawiz, 2020-08-09:
> I got the surface go as well.
> 
> @CornishRattler
> can you let us know what LTE WWAN card you have?
> 
> maybe its a driver issue for us @sohawiz

I found 'Qualcomm Atheros QCA61x4A Wireless Network Adapter #2' and 'Surface Mobile Broadband' on Device Manager(Network adapter).
I'm not sure which is for LTE Service, but I'll update drivers of both adapter

gyeo-ri, 2020-08-13:
> I got the surface go as well.
> 
> @CornishRattler
> can you let us know what LTE WWAN card you have?
> 
> maybe its a driver issue for us @sohawiz

Ahh, I've misunderstood  before.  You also have same problem, no? 
Actually I read a case that someone activate Mobile Data succesfully on Surface Go. But I couldn't got anything yet.  Do you have any update?  I tried a few time but it have never been activated

sebanc, 2020-08-29:
Microsoft Surface device suffer from a specific bug:
https://github.com/jakeday/linux-surface/issues/306

What you can try:

1) Identify your usb LTE  modem path:

`sudo find /sys/bus/usb/devices/ | grep rx_max`

2) Create a bootscript (replace the path with the one found in step 1) by running the 5 below commands:

`sudo mkdir -p /var/brunch/bootscripts`

`echo "echo 16383 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/rx_max" | sudo tee /var/brunch/bootscripts/lte.sh`

`echo "echo 16383 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`

`echo "echo 16384 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/rx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`

`echo "echo 16384 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`

Lupinforlord, 2020-09-17:
> Microsoft Surface devices suffer from a specific bug:
> [jakeday/linux-surface#306](https://github.com/jakeday/linux-surface/issues/306)
> 
> What you can try:
> 
> 1. Identify your usb LTE  modem path:
> 
> `sudo find /sys/bus/usb/devices/ | grep rx_max`
> 
> 1. Create a bootscript (replace the path with the one found in step 1) by running the 5 below commands:
> 
> `sudo mkdir -p /var/brunch/bootscripts`
> 
> `echo "echo 16383 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/rx_max" | sudo tee /var/brunch/bootscripts/lte.sh`
> 
> `echo "echo 16383 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`
> 
> `echo "echo 16384 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/rx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`
> 
> `echo "echo 16384 > /sys/bus/usb/devices/2-2/2-2:1.0/net/wwp0s20f0u2/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh`
> 
> `sudo chmod 0755 /var/brunch/bootscripts/lte.sh`

hi, I did `sudo find /sys/bus/usb/devices/ | grep rx_max` but nothing happens?

johnnylau34, 2020-09-26:
I am using Surface Go 2 LTE m3-8100y version and I got LTE working on Chrome os with below steps, followed by a reboot. Not sure if it works for you too (maybe you need to explore around /sys/bus/usb/devices folder and look for where you can find a net folder too if below does not work for you):

sudo mkdir -p /var/brunch/bootscripts

echo "echo 16383 > /sys/bus/usb/devices/2-3:1-0/net/wwan0/cdc_ncm/rx_max" | sudo tee /var/brunch/bootscripts/lte.sh

echo "echo 16383 > /sys/bus/usb/devices/2-3:1.0/net/wwan0/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh

echo "echo 16384 > /sys/bus/usb/devices/2-3:1.0/net/wwan0/cdc_ncm/rx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh

echo "echo 16384 > /sys/bus/usb/devices/2-3:1.0/net/wwan0/cdc_ncm/tx_max" | sudo tee -a /var/brunch/bootscripts/lte.sh

sudo chmod 0755 /var/brunch/bootscripts/lte.sh

[image]

Lupinforlord, 2020-09-27:
> I am using Surface Go 2 LTE m3-8100y version and I got LTE working on Chrome os with below steps, followed by a reboot. Not sure if it works for you too (maybe you need to explore around [...]

Hi, which bruch are you using?

johnnylau34, 2020-09-28:
chronos@localhost / $ cat /etc/brunch_version
Brunch r85 k4.19 20200919
chronos@localhost / $

johnnylau34, 2020-09-28:
> Hi, which bruch are you using?

chronos@localhost / $ cat /etc/brunch_version
Brunch r85 k4.19 20200919
chronos@localhost / $

Lupinforlord, 2020-10-08:
> chronos@localhost / $ cat /etc/brunch_version
> Brunch r85 k4.19 20200919
> chronos@localhost / $

Hi, did you do so under terminal in ChromeOS? I saw same files with excately same path as yours in linux mint and did the command but surely nothings works. And with an app called "total command" I can see there is only so far /sys/bus/usb/devices/2-3:1.0/net, nothing is in there, besides wired thing is I can't even see /2-3:1.0 with terminal in ChromeOS....

Lupinforlord, 2020-10-08:
> Hi, did you do so under terminal in ChromeOS? I saw same files with excately same path as yours in linux mint and did the command but surely nothings works. And with an app called "total command" I [...]

Succeeded! Turns out you need to:
1. Ctrl+shift+t (go to cross)
2.`shell`
3.`su` or `sudo su`
4.continue with the same commands above.

Tash254-ke, 2021-01-01:
I am not conversant with things like this but it looks like we're facing the same issue. I am using ThinkPad T460s and when I put a simcard chrome os doesn't recognize it.
I tried the commands above after a long hectic process of looking for sudo su password, and nothing works

crosh> shell
chronos@localhost / $ sudo su
Password: 
localhost / # sudo find /sys/bus/usb/devices/ | grep rx_max
localhost / # 
localhost / # sudo find /sys/bus/usb/devices/ | grep rx_max
localhost / # 
localhost / # sudo mkdir -p /var/brunch/bootscripts
localhost / # exit
exit
chronos@localhost / $ sudo su
Password: 
localhost / # sudo mkdir -p /var/brunch/bootscripts
localhost / # sudo find /sys/bus/usb/devices/ | grep rx_max
localhost / # 
The simcard doesn't have a pin, I have verified that using a phone, and I have different simcards from different telcos.
Anyone knows how I can find a solution?
On Lenovo's site, Huawei ME906S this is the WWAN card I think.
Thank you.

sebanc, 2021-01-01:
@Tash254-ke, your device needs a different configuration, you might want to try the "Linux support" part of the below article:
https://toreanderson.github.io/2017/07/31/huawei-me906s-hp-lt4132-linux-ipv6.html

Tash254-ke, 2021-01-13:
> @Tash254-ke, your device needs a different configuration, you might want to try the "Linux support" part of the below article:
> https://toreanderson.github.io/2017/07/31/huawei-me906s-hp-lt4132-linux-ipv6.html

I have been away for a while from this laptop. Today I checked this link and usb-devices doesn't even show the modem as per instructions. What could be the reason now?

Lupinforlord, 2021-03-28:
How is everyone's lye workinh? mine works occasionally and doesn't work sometimes after reboot, by cat lte.sh I know the file is unchanged so I have to do chmod again and reboot, then it works. Is there a way to fix this? @sebanc

Lupinforlord, 2021-03-28:
also found failure to connect to network while lte indicator still showing after waking from sleep. Currently with brunch r88

Lupinforlord, 2021-05-27:
@johnnylau34 
Hi, do you remember which rammus is yours? I tried to recreate system img, and tried lte.sh with rammus 90 and r90, r89. They all failed, and now trying rammus 89...

Lupinforlord, 2021-05-27:
> @johnnylau34
> Hi, do you remember which rammus is yours? I tried to recreate system img, and tried lte.sh with rammus 90 and r90, r89. They all failed, and now trying rammus 89...

Down. Turn out rammus 89 and r88 works.

Lupinforlord, 2021-05-28:
> @sebanc does lte.sh have conflicts with rc.local? I have rc.local activated for mobile broadband in linux mint and now neither of r90 or r89 can have lte running

I can confirm it's the problem with brunch, no conflicts between lte.sh and rc.local.

dustojnikhummer, 2022-01-26:
> > @sebanc does lte.sh have conflicts with rc.local? I have rc.local activated for mobile broadband in linux mint and now neither of r90 or r89 can have lte running
> 
> I can confirm it's the problem with brunch, no conflicts between lte.sh and rc.local.

Hey, almost a year late here. Did you fix this?  I have a Surface Go LTE

`sudo find /sys/bus/usb/devices/ | grep rx_max` returns nothing, BUT `/sys/bus/usb/devices/2-3:1.0/net/wwan0/` are there. Built the script you had above but I can't run it, even as a root (sudo su)

[image]

yasurok, 2022-12-03:
Hi. Just installed chromeos_15117.112.0_rammus on my Surface Pro LTE Advanced (2017, 5th gen).  After a lot of shots in the dark, the following gave me the access to the LTE modem on it. Now please someone advise me where I should place it so it will be run automatically at the right time. I am no IT expert and I have not touched *nix for a few decades.

```
#!/bin/bash

echo 16383 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/rx_max
echo 16383 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/tx_max

echo 16384 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/rx_max
echo 16384 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/tx_max

sudo restart modemmanager
```

I also have a feeling the `find`  command does not work as it should, which probably led to a lot of confusion in this thread.

yasurok, 2022-12-17:
[I was taught how to do it by the man himself](https://github.com/sebanc/brunch/discussions/1740#discussioncomment-4437310). :) Lucky me!

Gd7playz, 2023-01-14:
> Hi. Just installed chromeos_15117.112.0_rammus on my Surface Pro LTE Advanced (2017, 5th gen). Like many others, I had trouble getting the LTE modem even show up in the Settings.
> 
> After a lot of shots in the dark, the following finally gave me the access to the LTE modem on it, and I am happy to report I can get on the Internet through it.
> 
> Now please someone advise me where I should place it so it will be run automatically at the right time. I am no IT expert and I have not touched *nix for a few decades. :)
> 
> ```
> #!/bin/bash
> 
> echo 16383 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/rx_max
> echo 16383 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/tx_max
> 
> echo 16384 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/rx_max
> echo 16384 | sudo tee /sys/bus/usb/devices/2-2:1.0/net/wwan0/cdc_ncm/tx_max
> 
> sudo restart modemmanager
> ```
> 
> I also have a feeling the `find` command does not work as it should, which probably led to a lot of confusion in this thread.

Yo, I'm using a lenovo p51 with a sierra EM7455 sim card chip can you show me how you did yours

Caaruzo, 2023-03-28:
Hi there,

i got the same Problem on my Elitebook 1030 G3.
The mobile Option dont show up at all. I also cant find any rx_max or tx_max files.

```
lspci | grep LTE
6d:00.0 Wireless controller [0d40]: Intel Corporation XMM7360 LTE Advanced Modem (rev 01)

ls -la /sys/bus/pci/devices/0000\:6d\:00.0/
total 0
drwxr-xr-x.  4 root root    0 Mar 28 08:11 .
drwxr-xr-x. 10 root root    0 Mar 28 08:11 ..
-r--r--r--.  1 root root 4096 Mar 28 08:48 aer_dev_correctable
-r--r--r--.  1 root root 4096 Mar 28 08:48 aer_dev_fatal
-r--r--r--.  1 root root 4096 Mar 28 08:48 aer_dev_nonfatal
-r--r--r--.  1 root root 4096 Mar 28 08:48 ari_enabled
-rw-r--r--.  1 root root 4096 Mar 28 08:48 broken_parity_status
-r--r--r--.  1 root root 4096 Mar 28 08:11 class
-rw-r--r--.  1 root root 4096 Mar 28 08:12 config
-r--r--r--.  1 root root 4096 Mar 28 08:48 consistent_dma_mask_bits
-r--r--r--.  1 root root 4096 Mar 28 08:48 current_link_speed
-r--r--r--.  1 root root 4096 Mar 28 08:48 current_link_width
-rw-r--r--.  1 root root 4096 Mar 28 08:48 d3cold_allowed
-r--r--r--.  1 root root 4096 Mar 28 08:11 device
-r--r--r--.  1 root root 4096 Mar 28 08:48 dma_mask_bits
-rw-r--r--.  1 root root 4096 Mar 28 08:48 driver_override
-rw-r--r--.  1 root root 4096 Mar 28 08:48 enable
lrwxrwxrwx.  1 root root    0 Mar 28 08:48 firmware_node -> ../../../LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:4a/device:4b
-r--r--r--.  1 root root 4096 Mar 28 08:48 irq
drwxr-xr-x.  2 root root    0 Mar 28 08:11 link
-r--r--r--.  1 root root 4096 Mar 28 08:48 local_cpulist
-r--r--r--.  1 root root 4096 Mar 28 08:48 local_cpus
-r--r--r--.  1 root root 4096 Mar 28 08:48 max_link_speed
-r--r--r--.  1 root root 4096 Mar 28 08:48 max_link_width
-r--r--r--.  1 root root 4096 Mar 28 08:48 modalias
-rw-r--r--.  1 root root 4096 Mar 28 08:48 msi_bus
drwxr-xr-x.  2 root root    0 Mar 28 08:11 power
-r--r--r--.  1 root root 4096 Mar 28 08:48 power_state
--w--w----.  1 root root 4096 Mar 28 08:48 remove
--w-------.  1 root root 4096 Mar 28 08:48 rescan
--w-------.  1 root root 4096 Mar 28 08:48 reset
-rw-r--r--.  1 root root 4096 Mar 28 08:48 reset_method
-r--r--r--.  1 root root 4096 Mar 28 08:48 resource
-rw-------.  1 root root 4096 Mar 28 08:48 resource0
-rw-------.  1 root root 1024 Mar 28 08:48 resource2
-r--r--r--.  1 root root 4096 Mar 28 08:11 revision
lrwxrwxrwx.  1 root root    0 Mar 28 08:11 subsystem -> ../../../../bus/pci
-r--r--r--.  1 root root 4096 Mar 28 08:11 subsystem_device
-r--r--r--.  1 root root 4096 Mar 28 08:11 subsystem_vendor
-rw-r--r--.  1 root root 4096 Mar 28 08:11 uevent
-r--r--r--.  1 root root 4096 Mar 28 08:48 untrusted
-r--r--r--.  1 root root 4096 Mar 28 08:11 vendor
-r--r--r--.  1 root root 4096 Mar 28 08:48 waiting_for_supplier
```

Hope someone can help me please, get this to work.
Thx much.

Junenosuke, 2024-12-13:
Hi,
More than a year has passed.
The first thing I can say is that it is probably very difficult to use any product other than the Sierra EMM7430 as an LTE module in ChromeOS today. Or use a Surface as described above.
For example, the HP Elitebook 1030 G3 uses the Qualcomm® Snapdragon X12 as its LTE module, which is not recognized by ChromeOS.
I also tried to see if the EMM7430 was available for the HP Elite Dragonfly, but it was not even recognized in Windows Device Manager.
We have to wait for someone to develop a driver for these chipsets on ChromeOS or choose a device that has the EMM7430.
Unfortunately I don't have the time or ability to do that.

reinforchu, 2025-01-02:
Hi, This is the best news of the year. I have a Japanese Surface Go2 LTE and recently installed Chrome OS/Brunch. I found this issue because the mobile network was not working. I was able to get wwan0 to work with version r131. Below are the steps to reproduce.

[image]

- Access the virtual terminal in developer mode. (VT-2)

`Ctrl + Alt + fn + F2`

- Log in as root

The user name should be **root** and the default password should be blank.

- Identify the device name.

`sudo find /sys/ | grep x_max`

Check for device names that include "tx_max" and "rx_max" at the end.

In this case, the device name is "**/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/net/wwan0**".
[image]

- Rewrite the buffer size

```
echo 16383 | sudo tee /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/net/wwan0/cdc_ncm/rx_max
echo 16383 | sudo tee /sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3:1.0/net/wwan0/cdc_ncm/tx_max
```

- Restart the daemon

```
initctl stop modemmanager
initctl start modemmanager
```

**Not systemctl**.

- Return to the desktop

`Ctrl + Alt + fn + F1`

You can confirm that the mobile network is enabled in the settings screen.

Now I have successfully made the mobile network work. Have a great year everyone.