sebanc/brunch #444: Wifi support

GITCOM-CMD, 2020-08-02:
Not connecting to wifi, wifi status remains "**_Wi-Fi is turned off._**"
What I tried

1. added options=broadcom_wl **_didn't work_**
2. Used external usb Wifi dongle again **_didn't work_**
3. tried "sudo rfkill unblock all" **_didn't work_** 
4. Ethernet cable ->detected but status is always "**_Connecting_**"
5. Tried all network related options=broadcom_wl ,iwlwifi_backport,rtl8188eu,rtl8821ce,rtbth  **_didn't work_**

 

 

Hi please can you add wifi support for Sony device

  Computer Name: VAIO
 Computer Brand Name: Sony SVF14N13CXB

[Operating System]
Operating System: | Microsoft Windows 8  (x64) Build 9200
| UEFI Boot: | Present

**Network Adapter_** 
_Broadcom BCM43142 802.11b/g/n Wireless Network Adapter_

Device Name:  Broadcom BCM43142 802.11b/g/n Wireless Network Adapter
Original Device Name:  Broadcom BCM43142 802.11b/g/n Wireless Network Adapter
Device Class:  Unknown Network Adapter
 Revision ID: 1
PCI Address (Bus:Device:Function) Number: 2:0:0

 Driver Manufacturer: | Broadcom
Driver Description: | BCM43142 Wireless Network Adapter
Driver Provider: | Broadcom
Driver Version: | 6.30.59.132
Driver Date: | 08-Mar-2013

Laptop link
[https://www.cnet.com/products/sony-vaio-fit-14a-svf14n13cxb-14-core-i5-4200u-8-gb-ram-500-gb-hdd-qwerty/](https://www.cnet.com/products/sony-vaio-fit-14a-svf14n13cxb-14-core-i5-4200u-8-gb-ram-500-gb-hdd-qwerty/)

GITCOM-CMD, 2020-08-02:
Using dual boot with windows 8

GITCOM-CMD, 2020-08-02:
Wifi working fine on windows 8 but **not when booting from chrome os**

GITCOM-CMD, 2020-08-02:
Using **Brunch r83 k4.19 stable 20200708** with **rammus**

WesBosch, 2020-08-03:
I see that you are dualbooting. How are you making changes to grub, externally through windows or the grub menu, or through brunch using edit-grub-config? If you're editing using the edit-grub-config command, could you try adding options=broadcom_wl using the grub menu instead? (press e on the boot screen while ChromeOS is highlighted)

GITCOM-CMD, 2020-08-03:
@WesBosch Thanks Now wifi status is on but it is not showing(detecting) any wifi networks.

GITCOM-CMD, 2020-08-03:
@WesBosch Or fix usb wireless adapter detect, I inserted usb wifi adapter but it is not picking any usb port.

GITCOM-CMD, 2020-08-04:
?

WesBosch, 2020-08-04:
I only knew how to fix your initial grub issue. I don't have an answer for your USB problem, you'll have to wait for someone else who may know the answer. Be patient